gcormier/megadesk #2: BEKANT Power Supply Voltage

James-TR, 2018-03-31:
As you have noticed, the BEKANT power supply delivers 24V at idle (no load), which is under the name-plate voltage. The voltage increases to 35V as soon as a resistive load such as the motors is put across the power supply.

Would need to spec components to handle the increased voltage whilst the desk is moving. The gradual increase in voltage is apparently responsible for the soft-start.

gcormier, 2018-03-31:
The regulators I bought are good for input voltage up to 35v, although it might be worth it to look at a simple buck converter... 

I'll confirm the behavior on my side and make a note.

gcormier, 2018-03-31:
b63923b7e8339dc6f86580f9c62a1275d52c6570 references the issue with voltage for now. I've opened issue #3 to look into it further.

James-TR, 2018-04-01:
Thanks, it does seem interesting that they're overdriving the 24V motors at 35V... This may just be to get the extra speed.

closed 2018-04-15 by gcormier