gcormier/megadesk #5: Request: Image Output of schematic

jeffreypriebe, 2018-04-17:
It looks like you've upgraded the circuit from what is in https://github.com/trainman419/bekant/

(Judging by [your schematic snippet here](https://github.com/trainman419/bekant/issues/2#issuecomment-370231664) which shows diodes and a capacitor not in [trainman419](https://github.com/trainman419/bekant/blob/master/Bekant_schem.png) and also your BOM.)

It would be helpful if you had an image output of it. I'll try to find a tool to view [the relevant files](https://github.com/gcormier/megadesk/tree/master/PCB) but have been troubleshooting other issues with my attempted setup.

gcormier, 2018-04-17:
KiCad is the EDA used, it is FOSS - http://kicad-pcb.org/

There's also a few online viewers as well you can use. Here's a screenshot of the latest:

[image]

jeffreypriebe, 2018-04-18:
Thanks for solving this two ways!

closed 2018-04-18 by jeffreypriebe