gcormier/megadesk #7: Replace ATTINY841 with ESP-12F

tcprst, 2018-07-13:
Hey, first off I just want to say great work and that this looks pretty awesome. 

I just recently purchased an Bekant and was thinking about modifying the controller to add an MQTT interface implemented on an ESP8266.  I came across your project and I think I could almost just drop in a ESP-12F in place of the ATTINY841.  

Do you see any glaring issues with this approach, and do you have any advice for me as I go down this path?

Thanks!

gcormier, 2018-07-14:
Hey! If you mean as a general circuit then I don't think you will have any issues. I'm not super familiar with the ESP series - is it 100% arduino compatible? The only thing that could get complicated is the LIN interface.

Physically though, I don't think you will be able to re-use the IKEA case for the housing. It was already a tight fit with the TINY841 - the ESP-12F looks bigger and it might be a challenge to fit it inside the case. With that said, one interesting approach might be to 3D print a spacer to make the housing thicker - this would give you some more space and reduce the constraints around where you can put the capacitors.

tcprst, 2018-07-14:
It's pretty close to 100% compatible barring some hardware specific Atmel stuff.  The only thing I was worried about software wise was the HardwareSerial library but it appears to be functional on the ESP-12F.  I've compiled your code with an ESP as the target and it builds successfully, I'm just going to have to try it out.

Physically, the ESP-12F is quite a bit bigger and requires a 3.3v power supply and has 3.3v logic levels so I'll need to do some level shifting to the LIN Interface which will definitely take up some additional space. I really like that you reused the existing IKEA housing, but it will be a challenge to keep it in that same footprint.  If I manage to fit it all in the same footprint I could 3D print a spacer, that's an interesting idea.

Another thing I want to try is controlling multiple Bekant frames with one controller to make a larger sit/stand desk, but that will have to wait.

Thanks for the quick reply, and have fun with your megadesk!

gcormier, 2018-07-16:
I think the MCP2003B will run at 3.3V which should save you some components! Although you might then want to run the 3.3V voltage regulator off the 12V, as the heat to go from ~35V-3.3V will be even slightly more then down to 5V.

Do you have a scope or logic probe? That will go a long way to debugging :)

dizzy666, 2018-08-20:
Hey, great project. @tcprst do you already have the software running on an ESP?

closed 2018-09-03 by gcormier

luckydonald, 2019-12-29:
Hey you guys got something working on the ESP front here?

zoltanhosszu, 2020-05-26:
@luckydonald did you find something yourself? I’ve been struggling with an ESP8266 to get a proper LIN response, probably I’m doing something wrong :(