gcormier/megadesk #8: Spare boards/parts

gcormier, 2018-07-26:
I have enough parts for two units if anyone is interested - assembled or non.

APINAPOIKAMIES, 2018-09-15:
How much $ for one?

gcormier, 2018-09-15:
For the parts and PCB I'd say 15$ + shipping? Un-assembled.

rpmoore16, 2018-10-07:
Do you still have any parts? I'm interested in purchasing one unit un-assembled.

gcormier, 2018-10-08:
I do have 1 set left! Let me know your email address and we can work out the details.

gcormier, 2018-11-15:
Still have one set available!

AlexRoss-xx, 2018-12-04:
Is it still availible?

gcormier, 2018-12-04:
The last one is reserved for @rodcleveland, but I am thinking of putting these up for sale on tindie since there seems to be some demand! Is that something you would be interested in?

AlexRoss-xx, 2018-12-04:
yes. because it is impossible to buy some components in my country.

gcormier, 2018-12-08:
@AlexRoss-xx Hi Alex - please send me your email and we can discuss.

gcormier, 2018-12-12:
The spare boards are gone. I've ordered a new set of boards and parts and will be selling them via Tindie. https://www.tindie.com/products/gcormier/megadesk/

closed 2018-12-12 by gcormier

GeeWizWow, 2018-12-21:
Hey. Tindie is having a moment, won't accept e-mail addresses or quantities. I'd love to pick one up un-assembled

gcormier, 2018-12-21:
I've thought a bit about a parts-only version, but I"m unsure how to do it.. the capacitors are unmarked and would pose an issue... although, I could probably just dab them with a marker as there are only 3 different values.

I should have the last part (FPC header) in a week or two, then I can update the stock there and I'll try to rig up an parts-only option.

GeeWizWow, 2018-12-21:
Most places ship them in a little baggie with the value written on the bag with a sharpie / sticker or something. (Happy to skip the headers, have more than enough here)

gcormier, 2018-12-21:
Hmm a sticker with the part stuck on it as well might work. Even baggies would just be too big, and wasteful - these things are tiny! :)

GeeWizWow, 2018-12-21:
masking tape ftw

gcormier, 2018-12-21:
Good point! :) I'll post here when I get it up and running. Yeah I figure the kit would not come with pin headers - if someone doesn't already own hundreds of them... not a good sign for flashing it or assembly it successfully! :)

GeeWizWow, 2019-01-06:
Picked one up, thanks Greg