gcormier/megadesk #9: Ikea bekant

3bitru, 2019-01-19:
Hello

I see you have a deep understanding.

I purchased a table bekant on sale in Ikea, not working. Found the failed elements and replaced them.

Connected to the LIN bus.
If I connect slaves one by one they respond PID 8.
If I connect two slaves, PID 8 will still be recalled.

The markings on the boards 68002430 FW: 3.9 Rev.2 17 / 47L. Maybe they are the same?

If I connect one slave will he move?

3bitru, 2019-01-20:
Yes, I do not see the commands PID 9 on the bus. I assume that I do not have them in the slave firmware.

gcormier, 2019-01-20:
I'm not sure I understand the problem you're having. Slave firmware? Do you have custom hardware in the motors?

3bitru, 2019-01-20:
All hardware (firmware) is standard, but I do not see PID 9.

gcormier, 2019-01-20:
I would suspect one of your motors or controllers is broken or not working in some way then.

3bitru, 2019-01-20:
Я посмотрел внимательно.И я обнаружил что есть 0x08 (D0:54 D1:3 D2:0) и 0x09 (D0:0 D1:255 D2:0)

3bitru, 2019-01-20:
I looked carefully. And I discovered that there is 0x08 (D0: 54 D1: 3 D2: 0) and 0x09 (D0: 0 D1: 255 D2: 0)

gcormier, 2019-01-20:
Okay that is good, at least both are responding.

I'm not sure on the status though. You'd have to capture from the start to see what's happening and you can compare it with **Startup Preamble.logicdata**. I'm not sure what the status 0 means, I haven't seen that before.

3bitru, 2019-01-20:
The first success of the engines rotate if you press the button down, but if you press the button up, they do not rotate.

3bitru, 2019-01-20:
press the button down 
0x17 (D0: 0 D1: 0 D2: 2)  
0x08 (D0: 0 D1: 255 D2: 2)  
0x09 (D0: 0 D1: 255 D2: 2)
0x18 (D0: 0 D1: 0 D2: 189)

Not press the button
0x17 (D0: 0 D1: 0 D2: 0)  
0x08 (D0: 241 D1: 254 D2: 0)
0x09 (D0: 241 D1: 254 D2: 0)
0x18 (D0: 241 D1: 254 D2: 252)

press the button up
0x17 (D0: 0 D1: 0 D2: 2)  
0x08 (D0: 241 D1: 254 D2: 2)  
0x09 (D0: 241 D1: 254 D2: 2)
0x18 (D0: 241 D1: 254 D2: 188)

3bitru, 2019-01-20:
Perhaps the controller thinks the table is at the top?

gcormier, 2019-01-20:
You are using the IKEA controller? Or megadesk?

3bitru, 2019-01-20:
Ikea

3bitru, 2019-01-20:
This was the initialization mode. I installed the engines in the supports. I lowered it all the way down, the overcurrent triggered. And it all worked!

closed 2019-01-23 by gcormier