gcormier/megadesk #11: Bluetooth LE support

jeamland, 2019-01-29:
Is there room on the board for a Bluetooth LE module? I'd cheerfully write an app to control it if there were...

jottr, 2019-01-29:
Some people on HN already suggested to use an ESP8266 instead.   
@gcormier what are your thoughts about this?   
Generally, some kind of networking stack BLE or 802.11 would be very useful.

gcormier, 2019-01-29:
The main issue would be the physical size. The board space required for an ESP is quite big. Once I get the rush of orders sorted out I can take a look at board layout, but I don't think it's promising. Not only are we constrained in the X/Y, but also height - there are a few sections of the case that have ribs running across which limit where we can place taller components. I'm note sure if the ESP's height is slim enough to span those sections. Of course, snipping them away might be an option in some areas, would need more research!

gcormier, 2019-01-30:
[image]

This is what you're up against trying to get an ESP32 in there.

jeamland, 2019-01-30:
The hatched area is the problem? (I'm not sure how to read this image.)

jottr, 2019-01-30:
I'm confused. The ESP8266 comes on many different form factors. [This one](https://www.adafruit.com/product/2491) should fit the enclosure easily.

jottr, 2019-01-30:
Actually, why do you constrain yourself with the original enclosure in the first place? Wouldn't it be neater to have room for an LCD which could display the height and settings stored in memory? 
Also, additional buttons would be quite nice.

jeamland, 2019-01-30:
I'm fine with staying in the footprint of the original enclosure as that means less mucking around with extra parts. If I had BLE or WiFi then I could get my computer to display that stuff.

gcormier, 2019-01-30:
@jottr The ESP8266 sizing is more or less the same.

[image]

As mentioned in the readme, I didn't want a hacky looking enclosure that was 3D printed with ugly buttons. Getting custom membrane buttons is very expensive.

closed 2019-03-21 by gcormier