gcormier/megadesk #12: Serial Control

dbrgn, 2019-01-29:
It would be great if there would be a serial interface for controlling the desk lift from the computer!

I have started a similar project for an Actiforce desk lift: https://github.com/dbrgn/desklift/ It's not done yet, but the serial protocol used is the following:

```
- Baudrate 115'200
- Every command is a single signed byte
- Positive values move up, negative values move down
- The value must be multiplied by 0.1s to get the move duration
```

So for example when sending 0x13, the desk would move up for 1.9 seconds.

This should be quite simple to implement I in the firmware. The issue would probably be available pins.

It looks like the ATtiny841 has two USART. If I read your schematics correctly, USART0 is used by the LIN transceiver. USART1 is available at PA4/PA5, but those pins are shared with SCK/MISO from the programming header... But maybe this could be solved somehow, e.g. by automatically disconnecting the serial pins if there's power on the programming header?

dbrgn, 2019-01-29:
(By the way, this could simplify #11 since a BLE controller could be implemented externally with a separate uC. There are also ARM uCs available with integrated BLE, e.g. from Nordic.)

dbrgn, 2019-01-29:
Oh, and as an addendum 2: Does the Bekant table allow going to absolute positions? If yes, then of course the time based protocol doesn't make much sense and should be replaced by absolute positions. I used that protocol for the Actiforce desklift because it does not support absolute positions.

gcormier, 2019-01-29:
Yes, the commands specify the actual absolute position that the motors should go to.

It should be possible to simply use two of the programming pins with SoftwareSerial to accept commands. You can put resistors in line with both those lines. That would allow the programmer to still function if needed. Some creative work might be needed to access the header properly as the space is quite small in there! :)

dbrgn, 2019-01-29:
Ah, I forgot about SoftwareSerial :slightly_smiling_face:

MrFaul, 2019-05-24:
I have currently the option to buy some "new" Linak DL15 from a warehouse closing and I'm planning to upgrade my 2m² desk-monster.
However I would need to build a custom controller for that thing, so how far are you with reverse engineering the protocol?

gcormier, 2019-05-24:
@MrFaul , if this is in reference to @dbrgn reverse engineering the Actiforce desk, probably best done on his repository.

dbrgn, 2019-05-24:
I don't think it was in reference to the Actiforce desk, since those use a very different system for moving the table up and down.

gcormier, 2019-05-24:
IKEA uses Bosch motors.

Megadesk is a working product that can move the desk up/down.. the reverse engineering is done.

MrFaul, 2019-05-24:
Well Linak uses Bosch in this to supply IKEA, but I doubt they would change the interface.
As usual the Swede is trying to save every penny...
The bakant is basically a in-house bootleg version specially made for IKEA

drcursor, 2019-07-17:
Would really love to have either this or the Bluetooth option.
Any chance of a connected version of the board?

legomind, 2019-10-10:
I would love serial commands to hookup a esp32 and add wifi control. Because why not?

nutinshell, 2019-10-11:
Wi-Fi control is a lot more fun, please consider adding it :)

gcormier, 2019-10-11:
ESP32 will not fit inside the case which is not in-scope of this project, see #7

legomind, 2019-10-11:
I was thinking more as an external microcontroller

dbrgn, 2019-10-11:
If serial control is supported, you can easily add WiFi yourself using an external microcontroller.

gcormier, 2019-10-12:
TX1/RX1 can be accessed via MISO/SCK which are on the programming header. All that's needed is the code :) Notice the "help wanted" tag :)

westenberg, 2020-12-11:
This would be great (the bluetooth option and/or the external microcontroller option). Is there anybody who already started working on this?

tagno25, 2021-01-10:
I have created code that I think will work, but I do not have a controller I can test it on yet.  After it is tested I can submit a pull request.

[https://github.com/tagno25/megadesk](https://github.com/tagno25/megadesk)

gcormier, 2021-01-11:
Very cool! You should receive it shortly.

martingemme, 2021-01-13:
I discussed it with @gcormier but would also like to have other people's input!

From my perspective: USB would be the easiest way to get going since a good solid connection could be running through a single cable under the desk to the computer (no pairing required). Of course, it would limit us to being connected to a computer. But my assumption is that people have computers on their desks and just like the OP suggested, I'd want to program it through my computer.

What about my phone one could say? I'd argue against it, not only for pairing but also: say we want a timer, do you want to run the app in the background all day long? Or rather have your computer - on the desk when you work - have plugged in USB, that way the presets could start working the moment you start working? Also, anyone volunteers for said app? I think going wifi would be beneficial mostly - and only - for smartphones, in my view.

For me at least, a better first step would be to get USB and I'd volunteer to make a desktop app - unsure yet but I'd want a Task Tray app for its access/simplicity but am open to feedback - to control it. I just don't like the perspective of needing to pair the device. Also, if the argument is because of extra cost I would absolutely buy the extra cost to assure a good connection (but that's me). Also, if you have a desktop computer, you may not have wifi, what do you do then?

I'm really interested in people's feedback on my view.

fotiDim, 2021-01-13:
Personally I would like to have it independent of my desk computer. It is not always turned on. I would like to integrate it on Home Assistant and Alexa. My Home Assistant PC is far and out of Bluetooth range, let alone USB cable connection. 

I could add another RPI to act as the control unit of the desk but it seems like an overkill. My ideal solution would be wifi based with something like ESP8266, So interfacing with the ESP8266 over the serial pins seems to be the way to go for me.

gcormier, 2021-01-13:
My initial thoughts for hardware are that a 2nd module can be designed with the appropriate microcontroller (either USB or WiFi connectivity), and another LIN bus module. For someone with an existing megadesk, they would put the module in between the megadesk and the motor. It would communicate to the megadesk over the same 3 wires as the rest as well as receive power. The addon/companion would ask the main megadesk for current height, as well as send commands to raise or lower.

This means existing megadesk people can just add on a 2nd module if they want remote control.

The other option would be a serial interface, but it means potential soldering of the 6-pin header or serial wires to megadesk, and cutting of the plastic case to make room for a 2nd serial cable.

Of concern though as always is the ability for a desk to move with no human nearby. The companion could also have a physical on/off switch if desired. Or, another means could be confirming automated movements with the down arrow button. eg.

- Application requests companion to move to standing
- Companion requests move to this height to megadesk
- Megadesk beeps for 5 seconds indicating a move is about to occur
- User pushes the down button, which would either confirm permission to move (preferred), or to cancel the move. It could be up to the software to determine which mode to use so users can accept their own risk tolerance (ie. children/pets in the house)

westenberg, 2021-01-13:
I like both ideas but for me serial to my MacBook won't work as I connect disconnect it all the time and also would like to work on the desk without a computer on it.

As hardware I would prefer a simple ESP devboard like a Wemos D1 Mini Pro communication over USB serial or the serial communication pins.

Drilling a hole in the case of the Bekant controller isn't that big of a deal but it would be nice if I can order a megadesk controller that already has a connector installed so I don't have to modify or solder on the board.

martingemme, 2021-01-13:
@westenberg
I wish I knew more about controllers and so on.. I'm "only" a high level programmer.. For me USB would be ideal in terms of programming although someone would need to provide an SDK for it. If you guys rather like the Wifi route and can provide a sdk I would investigate testing the SDK later as well.

For the use cases described by @gcormier, they could be integrated to the micro-controller, but it'd be great to also have the option to let complete control (a dev-mode like feature) where I can experiment without limitations.

westenberg, 2021-01-13:
> @westenberg
> 
> I wish I knew more about controllers and so on.. I'm "only" a high level dev.. For me USB would be ideal in terms of programming although someone would need to provide an SDK for it. If you guys rather like the Wifi route and can provide a sdk I would investigate testing the SDK later as well.
> 
> 
> 
> For the use cases described by @gcormier, they could be integrated to the micro-controller, but it'd be great to also have the option to let complete control (a dev-mode like feature) where I can experiment without limitations.

It depends on the protocol what devboard you can use. For example a Wemos D1 Mini has SPI, UART, i2C. So as long as a standard protocol is used and (for me preferable) one of the above then it will work. Connection can be with a usb connector on the Megadesk but pin headers are also fine.

dbrgn, 2021-01-13:
The reason why I proposed serial control is that it's really simple to use from another microcontroller. So people could create WiFi remote-control boards, USB remote control boards, Bluetooth remote control boards, etc.

USB on the other hand is quite complex. And the current ATtiny most probably won't support it.

dbrgn, 2021-01-13:
Ah, and if you don't want a real add-on board, you can just run a long serial cable to your computer and plug it into a serial-to-USB adapter that you can get for 1$. That's essentially what this desk lift control software/hardware uses: https://github.com/dbrgn/desklift

tagno25, 2021-01-13:
The code I have written, but not tested, should let a second device increase and decrease the height, send the desk to a specified height, get the current height, write to memory locations, read memory locations, and load memory locations.

If i have to change the code too much then some feature may have to be removed, because my current code uses 100% of flash memory.

martingemme, 2021-01-13:
> Ah, and if you don't want a real add-on board, you can just run a long serial cable to your computer and plug it into a serial-to-USB adapter that you can get for 1$. That's essentially what this desk lift control software/hardware uses: https://github.com/dbrgn/desklift
Again I wish I knew more about all that (I only play with C# and tiny bit of C++). Could you recommend me some parts and docs on how to send commands to the megadesk?

> The code I have written, but not tested, should let a second device increase and decrease the height, send the desk to a specified height, get the current height, write to memory locations, read [...]
Seems like exactly what I'd want! When you get your megadesk to test, could you let me know what parts you have, and how to use your code? I'd try it/test it with you if you want!

dbrgn, 2021-01-13:
> Could you recommend me some parts and docs on how to send commands to the megadesk?

Right now the Megadesk does not support serial control, I think. So support for that would be added first, and you'd need to solder in 3 cables (RX, TX and GND).

But regarding serial in general, think of it as a socket, where you can write bytes and receive bytes. For example, here are the docs on the .NET SerialPort class: https://docs.microsoft.com/en-us/dotnet/api/system.io.ports.serialport?view=dotnet-plat-ext-5.0

And regarding parts: You can use a random USB-TTL-Adapter, here are some: https://de.aliexpress.com/wholesale?SearchText=ttl+usb You'll likely find others on Ebay or in your local hobby electronics store.

In any case, someone would need to implement support for this in the current firmware, using software serial (because the ATtiny doesn't have a dedicated hardware peripheral for this).

gcormier, 2021-01-13:
If people want an unfinished adapter, then definitely the easiest way is to buy any existing board from their favorite supplier, write up their own code there and control via a soldered serial header on the megadesk.

If the idea is to get a finished product, that's where a few more considerations come in to play. My original goals included not having the underside of the desk look like a bomb :) I would rather design something that existing users can take advantage of rather then require they buy a complete replacement, assuming the add-on can be made much cheaper.

@dbrgn The ATTiny841 has two hardware serial interfaces. 

[image]

Serial1 could be used with the existing programming header.

westenberg, 2021-01-13:
> If people want an unfinished adapter, then definitely the easiest way is to buy any existing board from their favorite supplier, write up their own code there and control via a soldered serial [...]

If only the headers are added to the pcb then the controller case can stay the same for general users and when you want to use it more advanced via serial then you only have to drill a small hole to run the serial cable through.

If it should communicate with external devices without any case changes then I think it should be done with BLE but I don't know if that fits in the case.

gcormier, 2021-01-13:
It's true, a simple USB to serial converter with 3 wires is all that would be needed if wired connectivity is all that's needed. It would just be one more wire under the desk.

WiFi or bluetooth becomes a bit more of a design challenge to do it sleek. Otherwise any dev board as well could be done and then the serial interface used.

martingem, 2021-01-13:
> So support for that would be added first, and you'd need to solder in 3 cables (RX, TX and GND).

It'd be amazing if it was added 👍 If anyone can do it (add the necessary code to handle this + how to flash..), I'll buy any hardware required to test it, as long as you can help me set it up.. Thanks a lot for the info @dbrgn !

tobru, 2021-01-13:
The work done by @tagno25 in https://github.com/tagno25/megadesk looks very promising and will hopefully land in the official megadesk firmware sooner or later.

Being able to control the megadesk via serial would fit my needs. I'd "just" have to flash the new firmware to my v5 megadesk, connect the serial ports via some soldered cables to an ESP32 and I can do "whatever I want" with it. Of course it needs some work on the ESP32, but that's part of the fun (at least for me).

gcormier, 2021-01-13:
The new ESP32-S2-MINI *might* fit inside the stock housing if I felt like an entire re-design. It would only increase the BOM cost by a dollar.

It wouldn't be for quite some time as only engineering samples are currently available, the timelines are a few months out before reliable stock is available.

tobru, 2021-01-13:
Wow, the ESP32-S2-MINI looks very cool! Of course that would be awesome to have it directly built-in in the megadesk. Until then I'm happy to hook up an external ESP32 to tinker around, I don't mind getting my hands dirty with electronics and some software tinkering, this sounds like a lot of fun and many things to learn.

gcormier, 2021-01-13:
Yes it's surely far out on the timeline, if it even happens.

Initially I used to sell the kits versus pre-assembled as I figured there would be a lot more people who want to hack or be more hands on. It was not the case, we are the few, many just want a plug and play option that works reliably :) So for the 20-30 people in the world who want to play, well, that's why we're here isn't it?

martingem, 2021-01-20:
I have just received it 👍 I'm super hyped to try it as is. Hopefully @tagno25 you'll receive your unit and you can make it work, I'd be willing to test!

tagno25, 2021-01-22:
I am having problems getting the serial port to output consistently valid data.
If I load the Arduino "ASCII table" example and set it to use Serial1, then it prints about ten good lines then about ten junk lines and repeats.

It seems that when running the attiny841 off of 5v, then the internal oscillator may not be stable.  There are two options that I can think of, either using an external oscillator or calibrating the internal oscillator.  Using an external oscillator would require a PCB redesign to move the button to different pins. I am not sure how to calibrate the internal oscillator, but the calibration would probably have to be adjusted for each individual microcontroller.

gcormier, 2021-01-22:
What baud rate are you running the serial at? The LIN bus operates at 19,200 without issues from the internal oscillator. If you can run it at something slower, even for sanity testing (eg 300bps). Otherwise I wonder if Serial0/Serial1 are fighting with each other?

I'll try to flash a board with your code this weekend to play around with it.

tagno25, 2021-01-22:
I was trying to run it at 9600.  I just tried 19200 with the board just printing out "Start" in the setup function and I am still getting some noise.  I will check with a different serial adapter when I get off of work.

It also seems I2C would take too much space to fit on the flash.

tagno25, 2021-01-23:
It seems that the serial adapter. or the wires for it, was having problems.  I tried a different adapter and everything seems to be fine.

The newest code that I committed to my fork has most of the serial functions working. The EEPROM related commands need more work and are currently commented out.

martingem, 2021-01-23:
> @tagno25 
> The newest code, that I have committed to my fork, has most of the serial functions working. The EEPROM related commands need more work and are currently commented out.
this is amazing! Do you think you could document the needed tools and how to flash?

With @dbrgn 's options (although ttl USB gives lots and lots of results... if I could get guidance.. 👍), I'd want to make sure I get the things needed and instructions to test, if possible!

gcormier, 2021-01-23:
> this is amazing! Do you think you could document the needed tools and how to flash?
>
> With @dbrgn 's options (although ttl USB gives lots and lots of results... if I could get guidance.. 👍), I'd want [...]

This is already covered - https://github.com/gcormier/megadesk#hacking-and-contributing

martingem, 2021-01-23:
> @gcormier 
> This is already covered

I think I'm far from understanding what's needed, looking only from that place. I agree that you documented parts of it, but I only see the programmer (with 10 pin) and the 6 pin connector listed (that would exclude the serial/usb part of it). I'm willing to learn, but I'm far from knowing where to start, knowing only Visual Studio Code out of all in this repo... I appreciate the tutorial included as well, I just believe it'd help people to have more information than just the environment used (VSCode) and part of the flashing chain (AVR programmer). Also, introducing a new feature - serial logging - would mean more docs, so I'd suggest documenting it. After that, if it's as simple as plugging each other (portion of 10 pin-6pin) great, I just don't see these instructions. Hopefully it makes sense :) Thanks

gcormier, 2021-01-23:
Compiling/flashing is rather straight forward and covered in many places. The docs link to this tutorial which is still relevant https://circuitdigest.com/microcontroller-projects/programming-arduino-using-platform-io-to-blink-an-led You could start by using the blink sketch, but beeping the built in buzzer instead of blinking a LED.

Serial itself is covered as well; 

- https://www.ladyada.net/learn/arduino/lesson4.html
- https://www.allaboutcircuits.com/technical-articles/csharp-windows-application-for-arduino/

As for the specific serial code implementation, yes, that will definitely need more discussion :)

tagno25, 2021-01-23:
The serial commands are 6 bytes long.

1. The first byte is the data start byte
2. The second byte is the command to execute
3. The third and fourth bytes are the height as a 16 bit number (int)
4. 
5. The fifth and sixth bytes are the button press number as a 16 bit number (int) to save/retrieve heights
6. 

---

The data start byte/character is currently "<"  

The command can currently be one of the following:
"+"    increase height by amount
"-"    decrease height by amount
"="    set height to value
"C"    Ask for current location
~~"W"    Write height to EEPROM location~~
~~"R"    Read height from EEPROM location~~
~~"L"    Load height from EEPROM location~~

- If the command is "+" or "-", then the 3/4 bytes are relative to the the current height and the 5/6 bytes are ignored.
- If the command is "=", then the 3/4 bytes are the absolute height value and the 5/6 bytes are ignored.
- If the command is "C" then the 3/4/5/6 bytes are ignored.  The current height is printed out as a "=" command.
- If the command is "W", then the 3/4 bytes are the absolute height value or can be set to 0 to save the current height.  The 5/6 bytes are the the number of presses to save/retrieve.
- If the command is "R", then the 3/4 bytes ignored.  The 5/6 bytes are the the number of presses to save/retrieve.  The data returned will be a "R" command with the 3/4 bytes being the saved height.
- If the command is "L", then the 3/4 bytes are ignored.  The 5/6 bytes are the the number of presses to save/retrieve.

---

- When pressing the up and down buttons the change is reported "+" or "-" commands.
- When loading via the buttons or command then the value is reported as an "L" command, with the 3/4 bytes being the height and the 5/6 bytes being the saved location
- When saving via the buttons or command then the value is reported as a "W" command, with the 3/4 bytes being the height and the 5/6 bytes being the saved location

gcormier, 2021-01-23:
Was hoping to have a nice code session - unfortunately has turned into wrangling the Pololu AVR programmer and begging it to program directly from PlatformIO, with no luck so far. I can admit defeat and dig up another AVR programmer, program it from a command line each time, or keep banging my head against it :) I'll go for a walk and decide later.

gcormier, 2021-01-23:
@tagno25  Do you want me to try and free up some memory in the base megadesk code? I've already found 80 bytes of Flash I can save and I can keep looking if that would be useful for you.

tagno25, 2021-01-24:
The way that the load command is implemented, the controller thinks that you have pressed the button repeatedly.  This implementation makes values 0, 1, and 16 unable to be used for memory locations.

gcormier, 2021-01-24:
I've merged into the serial branch. I"ve just applied some memory tweaks (mainly int's to byte's).

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.3% (used 232 bytes from 512 bytes)
Flash: [==========]  97.5% (used 7984 bytes from 8192 bytes)
```

We can save another 28 bytes of memory by changing `pushCount` from `int` to `byte`, but it has some effect on your serial code, mainly `push_addr` in some functions that do bit shifting.

tagno25, 2021-01-24:
I have converted `pushCount`, `push_addr`, and `memorySlot` to from `int` to `uint8_t`.  As well as changing `recvInProgress` from `bool` to `uint8_t`.
I have also changed all the `byte` to `uint8_t` to maintain a standard.

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.1% (used 231 bytes from 512 bytes)
Flash: [==========]  96.9% (used 7936 bytes from 8192 bytes)
```

`LIN_MOTOR_IDLE` could also be changed to `uint8_t` which would save 42 bytes.

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.1% (used 231 bytes from 512 bytes)
Flash: [==========]  96.4% (used 7894 bytes from 8192 bytes)
```

gcormier, 2021-01-24:
Okay I've changed `LIN_MOTOR_IDLE`

:) I actually change all the `uint8_t` to `byte`. All the same I guess :)

My plan is to let the serial branch "soak" for a bit and hopefully have people run it to comment. I'd like to run it for a month having it stable before merging to master.

I also need to consider if serial should be added with compile-time options. If the code base for other options grows based on some of the other features (motor recalibrate), that would explode our memory consumption and would mean we'd need to choose the features at compile time to fit on the attiny841. I can see if there any other avr's that have a similar price point but more memory.

gcormier, 2021-01-24:
The new AVR 1624 looks great - 16kb of program memory, 2kb RAM and 2 hardware UARTs. It's delay and not available yet, will be probably the summer until they're obtainable in a reliable qty, but maybe late spring I can get one to start a prototype.

dbrgn, 2021-01-25:
Oh wow, they still make new 8-bit AVR models? Nice, I thought those were deprecated in favor of the 32 bit SAMD series 🙂

gcormier, 2021-01-27:
Okay I've got the serial branch flashed on my desk, I'll run it for a few weeks before merging to master. Even then it may be disabled by default as I think 99% of folks won't use the serial function. Those who want to use it will need to open up the board and solder on a header anyways and get their hands dirty.

tagno25, 2021-02-08:
It seems that the output from holding the up and down buttons is incorrect.  The write commands are being called faster than the desk is able to move to the next position.  I am thinking that the only easy way to fix it, is to change the output to be a "=" command with the target height.

gcormier, 2021-03-02:
This is now in the newly made dev branch. I think I'll close this issue and if there's more work needed we can open another one or re-open this one if it makes more sense.

tagno25, 2021-05-29:
With the new ability to use both buttons for memory, should the load function be modified to work with the down button?

I can think of the following two options:
- new command (lowercase L?) to load the down button memory positions (+26 bytes)
- if the memory position is above DOWN_SLOT_START, and bothmode is enabled, then act like the down button was pressed (+24 bytes)

[2 more replies]

closed 2021-12-03 by gcormier