gcormier/megadesk #14: Additional Startup Preamble

gcormier, 2019-02-02:
Finding a new startup-handshake that deviates from the existing one.

Presents itself on the larger version of the BEKANT. Wondering if the motors are different for bigger table size so it's a different protocol. Could use feedback if anyone has Megadesk working on a big BEKANT.

gcormier, 2019-02-18:
Closing as it was figure out this was a different motor busy status...

closed 2019-02-18 by gcormier

llinfeng, 2020-03-17:
Just to confirm, the measurement for "bigger table size" is:
```
Length: 63 "
Width: 31 1/2 "
Min. height: 22 "
Max. height: 48 "
Max. load: 154 lb
```
Source: [Ikea's product page for BEKANT Desk sit/stand, black stained ash veneer, black, 63x31 1/2 "](https://www.ikea.com/us/en/p/bekant-desk-sit-stand-black-stained-ash-veneer-black-s59281814/)