gcormier/megadesk #16: Slight overshoot on recall

jpambrun, 2019-02-04:
First off, I have to say that having your desk learn new tricks after a couple of years is pretty damn neat.

It feel like my disk is slightly overshooting the destination by ~10-15 mm.
This is obvious if I go to position 1, the ask again for position (and again..) it will go back and forth ~20 mm instead of staying still.

Since I really have only two positions and the overshoot is always in the same direction I can account for it at the memory store phase.

I am fairly sure this can be fixed or improved.

Just a small nit picking as this is really awesome. Seriously, thanks!

gcormier, 2019-02-04:
Thanks! It was definitely a fun project :)

I have the same thing. If you play with some of the HYSTERESIS values, it might hit the amount a bit better.

The issue is that when the megadesk "stops", the motors actually coast a bit. So I guess I can spend some time to try and debug what the difference is between those two points and account for it when saving the memory. It's definitely something I've been meaning to look at, albeit low on the priority list.

jpambrun, 2019-02-04:
I understand the issue is related to inertia.

Without looking at the code I assume the hysteresis will only prevent movement when recalling the position that you are already supposed to be in, but that's really just a symptom of the inertia issue.

You could undershoot on purpose and adjust after the movement stops or we could figure out the average amount of coast and subtract that to the travel.. When I have some free time I will try to take a look at the implementation.

gcormier, 2019-02-04:
In an old build, I didn't have the acceleration/deceleration command in yet (PID18, command 135). So when you stopped moving, the desk would stop rather abruptly. If you had a glass of water full to the brim, it would probably spill a few drops.

As you mentioned, I bet the overshoot is constant, and if we can measure it, we can adjust the moving sequence.

This might also eliminate the need for the HYSTERESIS, as it was basically a means to stop the desk from moving "backwards" after it overshot the destination from the smooth movement... and then it overshot again, and again...

TjSnake, 2019-12-18:
Hi Greg, I've just received the bord with the cable and it works well. I was wondering if there is any update on this issue and what is necessary to update the device. Thanks.

gcormier, 2020-01-01:
Hi TjSnake, I haven't taken a look at this yet.

diabl0w, 2020-03-10:
i also purchased the board and have this issue. If I am at a recalled memory position, and then recall the same memory position, the desk will move almost exactly 0.5" either up or down, and then if I recall it again, it will move the opposite up/down the same 0.5". Can there be some code that if moving from a lower position, will tell the desk to stop 0.5" lower and if currently at a higher position, then stop at 0.5" higher than the destination?

diabl0w, 2020-04-04:
I took some more precise measurements... assuming a saved height of 30 inches we have 2 things that can occur:

>  moving down from a starting height >30
>  * table lands at 29 9/16 inches (-7/16") 
> moving up from a starting height of <30
>  * table lands at 30 7/16 inches (+7/16")

this is completely consistent so it should be easy to compensate for in code

gcormier, 2020-04-04:
https://github.com/gcormier/megadesk/blob/8e9bb143c54dffb115661209b8b5397b57bedc2e/Code/src/megadesk.cpp#L224

I can't help but wondering if when we load the memory for the targetHeight, if we add the HYSTERESIS value.

````
if targetHeight > currentHeight
targetHeight -= HYSTERESIS

if targetHeight < currentHeight
targetHeight += HYSTERESIS;
````

Up for trying it out? Above assumes it's overshooting, if it's undershooting, flip the +/-.

diabl0w, 2020-04-04:
> https://github.com/gcormier/megadesk/blob/8e9bb143c54dffb115661209b8b5397b57bedc2e/Code/src/megadesk.cpp#L224
>
> I can't help but wondering if when we load the memory for the targetHeight, if we add [...]

I can try right now, but can you give me a better detail of where exactly to put those lines?

diabl0w, 2020-04-05:
it did help, it cut the offset down from ~0.5 inches to ~0.25 inches... but looking at the code and the math, it seems like all this is doing is increasing the HYSTERISIS value by a multiple of two, because you are changing the targetHeight by the HYSTERISIS value and then later comparing the currentheight and targetheight by the HYSTERISIS value again... I will test removing the change and just increasing HYSTERISIS

gcormier, 2020-04-05:
Thanks. So I've made a new branch "overshoot" that incorporates your 120 hysteresis as well as a coasting value.

This brought it to within 2mm recall accuracy, in either direction, which is pretty good.

Can you verify this branch works for you well?

diabl0w, 2020-04-05:
> Thanks. So I've made a new branch "overshoot" that incorporates your 120 hysteresis as well as a coasting value.
> 
> This brought it to within 2mm recall accuracy, in either direction, which is pretty good.
> 
> Can you verify this branch works for you well?

can you explain coast? I think if you remove that and just leave hysterisis at 120 then that will eliminate your 2mm issue

gcormier, 2020-04-05:
I found that with only HYSTERESIS set to 120, it would still overshoot/undershoot. It could be a difference in the desk motors - do you have the bigger or the larger BEKANT?

diabl0w, 2020-04-05:
> I found that with only HYSTERESIS set to 120, it would still overshoot/undershoot. It could be a difference in the desk motors - do you have the bigger or the larger BEKANT?

I am honestly not sure because I have never seen another bekant desk, I believe I have the larger one because I have to press the 16x up button for it to work. That being said, at least in my experience, whatever hysterisis value you do set, it should always land in the same offset... meaning the wrong hystersis value wont make it over shoot one time and then undershoot another time. also if it overshoots, it wont be by 2mm one time and 5mm another time. this is what made it easy to debug because I just kept increasing hystersis until it got closer and closer and then eventually perfect. and it will now be perfect every time. I guess thats also why I dont see a point in having a coast function.

either 120 will work for everyone (which apparently it wont), or we can offer setting hystersis in the down button functions, or users can just compile their own. but if you have coast at 17 and hystersis at 120, I bet you will get the same result if you just set hysterisis to 137 and remove coast. correct me if im wrong

gcormier, 2020-04-05:
I'll try 137 as suggested to see what the results are. The issue was that moving to a preset from below that height, versus moving to the same preset from above the height had different results.

diabl0w, 2020-04-05:
> I'll try 137 as suggested to see what the results are. The issue was that moving to a preset from below that height, versus moving to the same preset from above the height had different results.

maybe if you have a lot of weight on the desk it would cause overshoot on move down to preset and undershoot on move up to preset, just one of those weird things that can be overlooked

gcormier, 2020-04-05:
Looks like 137 works well enough on my desk. I have the 120cm (47") wide version - which do you have?

diabl0w, 2020-04-05:
> Looks like 137 works well enough on my desk. I have the 120cm (47") wide version - which do you have?

I have the 63" wide version... and yeah I believe thats expected, the math in the code works out to stop movement (start coasting) at the same place for whether you have 137 or 120+17

there is really no easy way that I see to seperate hysterisis and coast without requring a much bigger change in code. 

if we did seperate the two, I believe coast would have the much larger value.

I just dont think its worth the effort, the only side effect I see of setting such a large hysterisis value is that if you move to a preset and then make a small micro adjustment and then press to go back to the preset, the desk wont move back because its still within the accepted hysterisis range. I really doubt this is going to be a scenario anyone would run into anyway

gcormier, 2020-04-05:
Does 137 work well on your desk? With a laser measure I'm getting within 1-2mm.

diabl0w, 2020-04-05:
> Does 137 work well on your desk? With a laser measure I'm getting within 1-2mm.
> 
> If there's a big enough difference I can save the HYSTERESIS value in the EEPROM, same as I do for motor commands for the different sizes.

I dont have the desk apart to try and flash 137, I packed everything away after setting 120... I double checked and I am also within 1-2mm, measuring with a meterstick

gcormier, 2020-04-05:
I think this solves it - thanks for your help!

closed 2020-04-05 by gcormier