gcormier/megadesk #18: Desk doesn't move on button press

wereHamster, 2019-05-28:
I hear the four beeps during startup, and both testing modes also work as expected. But when I press the buttons the desk doesn't move. I don't have an oscilloscope unfortunately, so I can't tell you what's happening on the LIN bus.

Do you have any idea what could cause this and how we could further investigate?

gcormier, 2019-05-28:
What size desk do you have?

I would suggest the following

1 - Unplug the megadesk controller.
2 - Power on the desk.
3 - Plug in megadesk
4 - See if it works

I've noticed sometimes after a power failure in the house, unplugging and replugging just the controller repairs it.

gcormier, 2019-06-06:
Closing due to no activity.

closed 2019-06-06 by gcormier

maximemcduff, 2020-05-09:
Hi @gcormier, I have one of your preassembly part. Notting move too. I have the large one but i dont heart any sound after toggle up button 16x. I tried also factory reset with down button at startup, and try test mode with up button at startup. I tried what you tell in this issues. I heard 4 bip at startup like describe in install process. Thanks for your help Max

gcormier, 2020-05-09:
When you toggle up 16x, you should hear a series of tones afterwards
confirming the mode has changed. Did you hear that?