gcormier/megadesk #20: Memory function on both buttons

TjSnake, 2019-11-26:
Hi,
this is really nice project. I'm just wondering why the device doesn't use it's down button for memory same way as up button. It would reduce the number of presses for particular memory 2 fold. Having several sitting positions on down button and several standing positions on up button would help the use case by a lot.

Thanks

gcormier, 2019-11-26:
I assumed most would only have a handful of memory positions. It is definitely possible, and would be rather simple for someone wanted to contribute to submit a pull request ;)

TjSnake, 2019-12-02:
Yea, I was hoping that you as the author can do it. Since it would be quite easy for you. I'm in no position to do it myself.

scubafly, 2020-05-04:
I love this idea! I’m a coder myself, ordered the assembled version from tindie because I have little soldering experience. 
I would love to write some code for this, but have no idea how to get it on the controller.

gcormier, 2020-05-04:
Hey Jack I just filled your order last night but haven't put it in the box yet. I'll swap it out for one with a programming header that is soldered on if you're interested.

You would need an AVR ISP. There's a million cheap clones on Amazon, or slightly better ones from SparkFun/AdaFruit. Plug it in to your PC (if you're on Windows, probably spend a bit of time messing with drivers and [Zadig](https://zadig.akeo.ie/), and flash it with platformio.

scubafly, 2020-05-04:
That would be great! My main machine is a mac, but I have PC that runs ubuntu/windows as well.

gcormier, 2020-05-04:
If you're on Mac even better, should work out of the box with little/no fiddling I'd imagine. I use Linux and it works out of the box.

closed 2021-05-26 by gcormier