gcormier/megadesk #21: IDÅSEN support

gcormier, 2020-01-02:
Ikea has released a new motorized standing desk, the IDÅSEN.

Please thumbs-up this issue if support for that desk is of interest to you. It would require me investing in the purchase of a desk to reverse engineer, so there's significant costs to doing so.

leoluk, 2020-01-03:
The new desk has bluetooth support, so it's probably easier to reverse engineer that.

gcormier, 2020-01-04:
Ah interesting. This would very much be a software project I imagine, rather than hardware.

HikariMisako, 2020-06-07:
Some comments on reddit suggest that the Linak DPG1C controller is a plug and play option that adds memory and a display for the Idasen. Even a review of the controller on Amazon suggests the same thing, so while it's more expensive at 100$ it could be a solution for people hoping to upgrade their Idasen desk.

closed 2020-08-30 by gcormier