gcormier/megadesk #22: Compiles but doesnt run

diabl0w, 2020-03-11:
The latest master compiles and uploads in vscode, but the board doesnt beep or control the desk

diabl0w, 2020-03-11:
to add to this, could you tell me what the fuse values should be from avrdude? and also, are you shipping the devices with a bootloader?

I mostly ask because I may have inadvertently altered some of those things when trying to reprogram

gcormier, 2020-03-11:
Hi, you can find the fuses at the bottom of the readme, or in the [platformio.ini](https://github.com/gcormier/megadesk/blob/2e4ba49c9e9d9f41f0cf8ddcea00d7a83b18688b/Code/platformio.ini#L5). We only set the lfuse, the efuse and hfuse are default values.

If you aren't hearing any beeps then yes, likely fuses are incorrect.

diabl0w, 2020-03-11:
> Hi, you can find the fuses at the bottom of the readme, or in the [platformio.ini](https://github.com/gcormier/megadesk/blob/2e4ba49c9e9d9f41f0cf8ddcea00d7a83b18688b/Code/platformio.ini#L5). We only [...]

I only see the lfuse on both the readme and the platformio file, which I properly set... I assume the other fuses should just stay default but the problem is that I think they may have gotten changed and finding the default fuse values for the tiny841 hasnt had any luck. and all the online fuse calculators dont seem to have the 841 as an option

gcormier, 2020-03-11:
Default fuses are in the [datasheets](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8495-8-bit-AVR-Microcontrollers-ATtiny441-ATtiny841_Datasheet.pdf).

efuse is 0xFF
hfuse is 0xDF
lfuse is 0x62

diabl0w, 2020-03-11:
thanks, I just tested with the proper fuse values, I actually just found them in your "flash.bat" script about 2 minutes ago. The master src is still not working properly though. If flash your precompiled binary from the releases page, the board works. Compiling from source is whats not working

gcormier, 2020-03-11:
I just flashed a batch of these a couple weeks ago, trying to remember if I
used PIO or not. I might have used the precompiled release. I'll see if I
can get some time in the next few weeks to flash one directly from PIO and
see if it's working.

diabl0w, 2020-03-11:
thanks, i am just trying to get more accurate stop points when recalling a memory position, so i wanted to play with maybe the hysterisis value... ill check back for updates

gcormier, 2020-04-04:
Confirmed, migrating to platform.io broke something. I'll take a look.

gcormier, 2020-04-04:
It was a pin mapping issue in moving from ATTinyCore to whatever framework platform.io uses. Compiled, flashed and got some beeps!

commit 8e9bb143 (2020-04-04): Fixes #22 and add all fuses explicitly
closed 2020-04-04 by gcormier