gcormier/megadesk #23: Set custom max/min value

conrad784, 2020-05-13:
First of all, nice work I really missed the storing function on my desk :)
I've got an enhancement proposition. I got the case of a big container under my desk. Usually that is no problem because I know this, but if another person tries a lower position, it would be nice to store a minimum and maximum position which should not be run over. E.g. by pressing 20 and 30 times or something tedious like this (child safety lock like).

gcormier, 2020-05-13:
That's an interesting use case! :) But definitely doable - the min/max are set as constants presently, but they could be loaded from EEPROM. Button pushes could be used to set these values, 20x for lower limit, 22x for upper.

(I do look back and wish I'd limited the memory to maybe 8 positions so these other functions would be less annoying to access, but with many units out there in the wild I would rather be consistent)

conrad784, 2020-05-13:
I am very inexperienced with AVR programming, thats why I bought it assembled some time ago on tindie, but I'd like to put some time into adding this feature. I read in other issues you are recommending some hardware https://github.com/gcormier/megadesk/issues/20#issuecomment-623435948
Could you provide a "development quickstart" in your Readme, something like:
1. Supported way is using this USB-Dongle.
2. Add those headers to your unit.
3. connect pins like this (Everytime I switch TX and RX around in my serial interface history ;))
4. Compile with `make all`.
5. Program the unit with `avrdude -c ...`.

gcormier, 2020-05-13:
Sure, I will put some basic links together - there's a lot out there so I'll try to find some good information.

gcormier, 2020-05-24:
25a2157ff1ae398b11c9f17db43e2e7de8a55d3a adds some documentation on programming at the bottom of the readme now visible at the front page. I'm going to hide these comments - if you have further questions on programming, happy to help, just open a new issue so we keep this one dedicated to max/min values as I like the idea!

djalexz, 2020-10-06:
Hi gcormier,

I got my controller today.
Thank you very much for the fast delivery.
And it's really a great part.

Could you still put in the limit for up and down?
I have a small cabinet under my desk.
Without limit it would be broken :-)

Thanks a lot

gcormier, 2020-10-07:
Hello! It is not yet implemented. Firmware values could be changed if you are able to re-flash the controller.

djalexz, 2020-10-07:
I have this Converter https://www.amazon.de/gp/product/B01CYBHM26/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Can i use this to flash the Firmware?
And what value must be changed for the min value?

THX

gcormier, 2020-10-07:
No, that is merely a serial converter, which will not work for AVR programming. You can find more info at https://github.com/gcormier/megadesk#programming

gcormier, 2021-03-02:
This is now also in the newly created dev branch.

I made a small change though - when we do an init, we will want the default min/max to be the biggest range possible.

```
void initAndReadEEPROM(bool force)
{
  int a = EEPROM.read(0);
  int b = EEPROM.read(1);

  if ((a != 18 && b != 13) || force)
  {
    for (unsigned int index = 0; index < EEPROM.length(); index++)
      EEPROM.write(index, 0);
    // Store unique values
    EEPROM.write(0, 18);
    EEPROM.write(1, 13);

    #ifdef MINMAX
    // reset max/min height
    EEPROM.put(40, DANGER_MIN_HEIGHT);
    EEPROM.put(44, DANGER_MAX_HEIGHT);
    #endif
  }
    #ifdef MINMAX
    EEPROM.get(40, minHeight);
    EEPROM.get(44, maxHeight);
    #endif
}
```

closed 2021-05-26 by gcormier