gcormier/megadesk #24: Martela Alku S&S support (Rol Ergo system)

lauritk, 2020-06-06:
I have two spare legs from four legged Martela Alku S&S table which has Rol Ergo System similar to Bekant. I have recorded LIN bus logic and it looks very similar to your findings with the Ikea desk. I am trying to make megadesk work with my legs due to fact my friend has the working controller and psu :). If you're interested I can dump recordings somewhere to make them available?

Btw, did you implement reset functionality? I didn't find recordings for the reset function. In my desk reset is done by pressing both, up and down, buttons 8 seconds and then lowering table to bottom. I have recordings for reset too.

gcormier, 2020-06-07:
I didn't even know there was a reset functionality on the BEKANT; what does it do?

Sure if you want to make a subfolder in "Reverse engineering" and add one for this desk and do a pull request I can add it.

lauritk, 2020-06-08:
Interesting, it is not stated in the manual of Ikea Bekant, but according to [this](https://www.reddit.com/r/IKEA/comments/7nrp2w/debugging_the_bekant_sitstand_desk/ reset/recalibration) Reddit  Bekant should have reset/recalibartion too.

I am not completely sure what the reset does, but I think it will recalibrate the legs lowest position values. Because my Martela supports multiple legs and controllers (maybe), it might do something else too on the MCU. Unfortunately I don't have dumps of the firmware. I believe that Martela and Ikea tables base on Rol Ergo's iDrive system.

I will put data available after pruning etc.

gcormier, 2020-07-06:
#25

gcormier, 2020-07-06:
So the pictures looks identical to everything. This company built the IKEA Bekant so it's not surprising they've building for other companies.

Have you opened up the controller box?

lauritk, 2020-07-07:
>So the pictures looks identical to everything. This company built the IKEA Bekant so it's not surprising they've building for other companies.

To my understanding they are solely subcontracting company.

> Have you opened up the controller box?

Unfortunately I have not, did not want to destroy friends unit :). But I am guessing there is no big suprises on the pcb compared to Bekant. The LIN protocol seems very similar, but I still have to make some deeper analyses.

gcormier, 2020-07-08:
Okay, so the protocol has very similar patterns, but it is not the same.

- It repeats a sequence of dumb packets at the start, but as opposed to repeating, it also has some that increase from PID 1 to PID 7
- PID18 appears to be the master command, same as megadesk, but the commands (D2) are different.

So in short, definitely possible to hack up. The initialization handshake was by far the hardest part to decode for megadesk. You can bypass this if you can plug in the original controller for the initialize. Unfortunately a simple replay probably won't work.

By sending commands with the various legs unplugged you can discover what the PID is for each leg. Then you can start to make sense what is going back and forth. Since the motors are the same I imagine the structure of the commands and control loop are as well.

- Idle is just looping the same commands and status commands.
- Movement will have a pre-move command, then move commands, and then a finish-move command. If I recall, I think the move command can be done alone, it will just not have smooth acceleration or deceleration without the other commands.

gcormier, 2020-07-10:
Thanks for these notes btw, the reset DOES apply to the BEKANT and it helped solve an issue for a customer where the desk was not responding!

lauritk, 2020-07-10:
Thank you for viewing the recordings and hints!

> So in short, definitely possible to hack up. The initialization handshake was by far the hardest part to decode for megadesk. You can bypass this if you can plug in the original controller for the initialize. Unfortunately a simple replay probably won't work.

That's a bummer. I was hoping that the initialization would have been simpler, but after seeing that it does something more than I expected, that was my fear :smiley:. Initial plan was not to buy expensive (65 euros incl. postage) original handset, but I have now ordered it anyways to make tinkering bit more flexible. This seems to look like a project now :smile:.

> Thanks for these notes btw, the reset DOES apply to the BEKANT and it helped solve an issue for a customer where the desk was not responding!

Great! :partying_face:

gcormier, 2021-05-27:
I'm going to close this issue as it's likely not directly under megadesk goals.. but if you ever need to reopen or start new discussions on reverse engineering the Martela Alku, please feel free! :)

closed 2021-05-27 by gcormier