gcormier/megadesk #26: Up/Down buttons responding to input but desk isn't moving

levindixon, 2020-07-08:
👋 @gcormier 

I've just installed the `megadesk v4` and I hear the 4 tones on boot but the up/down buttons aren't moving the desk at all. They do respond to input though, I booted into the test mode and confirmed that the board is receiving a signal from both the up and down buttons.

I've tried the following to no avail: 

```
- Try unplugging the desk from the wall and plugging it back in (With megadesk connected)
- Alternatively, try disconnecting megadesk, removing power from the desk, adding power, and then re-connecting megadesk
```

Any suggestions?

p.s. was [this](https://www.youtube.com/watch?v=nwoIJxDRu8M) the inspiration for the name? 😄

gcormier, 2020-07-08:
It most definitely was the inspiration :)

Hmm looks like you've done most of your homework. Is the safety pin fully inserted?

levindixon, 2020-07-08:
> It most definitely was the inspiration :)

Amazing!

> Hmm looks like you've done most of your homework. Is the safety pin fully inserted?

Yeah, otherwise the board doesn't receive the up/down inputs at all...I left the pin in and swapped to the OEM board and the desk moves up and down normally 🤔

gcormier, 2020-07-08:
Hmm odd indeed. If you leave the desk powered on, and hot swap the boards, any luck?

whydna, 2020-07-09:
Same thing is happening here. It was working yesterday.

I can hear the click when I hold the UP button but it doesn't actually move.

jchambers2012, 2020-07-09:
what size desk do you have?
Small is 120cm x 80cm (47 1/4" x 31 1/2")
Default firmware, should work out of the box
Large is 160cm x 80cm (63" x 31 1/2")
Requires the 16x toggle - These motors report their "idle" status value as 0 instead of 96, which can cause the buttons to not respond. Pressing the memory (UP) button 16 times will play a tone and toggle the value (saved in EEPROM for persistence) that it will use.

levindixon, 2020-07-09:
> Requires the 16x toggle - These motors report their "idle" status value as 0 instead of 96, which can cause the buttons to not respond. Pressing the memory (UP) button 16 times will play a tone and toggle the value (saved in EEPROM for persistence) that it will use.

_**bingo**_, works! Thanks @jchambers2012

closed 2020-08-30 by gcormier