gcormier/megadesk #28: Add some helpful tips

dandantsui, 2020-08-16:
Hey! I just wanted to say thanks for making this and I have built two of these for my desks and they have been working great! I hand soldered everything onto the board successfully without flux or a hot air rework station!

With my longer BEKANT (160cm), I was not required to use the 16x toggle. I suspect this is due to my age of my desk... (2015/34).

A note with soldering the diodes, I found that in both instances, the D2 top pad would lift off -- I successfully used the length of the leg to jump to R6 bottom pad -- perhaps a JLCPCB issue? -- happy to provide a photo if you want.

Some flashing tips for others, I used a STM32 blue pill and had this flashed on it. [http://amitesh-singh.github.io/stm32/2017/05/21/FASTUSBasp-programmer-avr.html](url) then followed with avrdude with usbasp-clone as the programmer and "-B 3" flag to slow down the internal clock. I was stuck with flashing and every time it would result in "invalid device signature" and I thought my soldering was bad!!

Also, inside the kicad schematic, I found that the programming pins were flipped on the v4 and don't reflect the actual Gerber .zip file -- took me a while to notice!! but no harm was done to the ATTiny841 when I tried the pinout from the schematic.

Hope this info helps for others that are building this!

gcormier, 2020-08-16:
Nice thanks for reporting in! I'm glad you got things going!

One of the design goals was to make it relatively easy to hand solder if needed - even without hotair. Mind you, life is easier with hot air!

The diode pads are small indeed. I have not had any issues with lifting off, but it could of course depend on many things.

gcormier, 2020-11-14:
Note that v5 of the board now has a "regular" programming header.

closed 2021-02-11 by gcormier