gcormier/megadesk #30: Emulate recalibrate/reset the motors function?

RonMacNeil, 2020-09-18:
Would it be possible to emulate the recalibrate/reset motor process from the original BEKANT controller into Megadesk thereby avoiding the need to swap in the original controller to address this occasional issue?  Swapping in the original controller isn't trivial in my case...

I suspect that I'm facing this issue since I'm often not getting a good sync when using the larger desk (variant mode).  I would need to power down and then up and then toggle the variant mode twice to get it working again.

Thank you in advance,
Ron

gcormier, 2020-09-23:
It might be. I'd need to get everything wire up again to reverse engineer the protocol used to reset the motors.

zoltanhosszu, 2020-12-21:
This would come in handy for me as well, as I broke the original controller so I'm solely relying on MegaDesk :)

gcormier, 2021-05-27:
@philwmcdonald I've merged this into the dev branch, but the trigger to activate it is not there yet. We had discussed some other options, not sure if you think this would be a good candidate for push/hold of both buttons simultaneously?

```
// Temporary use pushcount 11 for recalibrate
    if (pushCount == 11)
    {
        state = State::STARTING_RECAL;
    } else 
```

philwmcdonald, 2021-05-27:
Maybe if both buttons are pressed during boot would be a good trigger. It's such a rarely used feature.

gcormier, 2021-05-27:
I dislike the power-on functions as it can be tricky to do this as one person :)

Since we have min/max setting, I would propose combo single button presses, and anything over 10x is "parameter" territory.

11x - Set minHeight
12x - Set maxHeight
14x - Recalibrate
16x - no longer used, do not use in future since old versions used it for idle status

philwmcdonald, 2021-05-27:
I hadn't considered how difficult it is to do the power-on functions - harder still if you have to hold both. 
I has using avrdude to reset which of course is easy and not an option for the regular user.

Ok, I'm convinced. I think this makes sense. 
I haven't pulled from your repo in a while, I'll merge from dev and put something together. If single and both buttons operation is a run-time parameter there are some optimizations I can make in the handling of buttons...

philwmcdonald, 2021-05-28:
PR #66 is up now with these ideas implemented.

gcormier, 2021-06-04:
Closing as dev now has recalibrate.

closed 2021-06-04 by gcormier