gcormier/megadesk #31: Megadesk sometimes not responsive

JanBe, 2020-10-19:
I have been using Megadesk for the better part of a year (bought November 2019, fully assembled plug and play) and it has mostly been great.

However sometimes I have the issue that the table is completely unresponsive to any button inputs. This didn't happen at all in the first months but now it begins to happen quite regularly. The fix is to unplug the power from the table and therefore force a restart of Megadesk, after that it is working fine again.

Unfortunately I could not find a pattern so far. Recently I came back to my desk after a 3 week holiday and Megadesk was still working fine. They day after however it was crashed again, after not having been used the 3 weeks before. This leads me to believe that is has something to do with input/ usage, not only time that passes.

I will try to identify a pattern as to when this happens. Do you maybe have an idea what the issue could be?

gcormier, 2020-10-19:
Question - are you leaving the megadesk powered on 24/7, or are you turning it on/off with a powerbar with other peripherals on your desk?

There is a handshake edge case that I haven't trapped. I just finished assembling and testing 30. Of the 30, one of them failed the initial plug-in. Resetting the power solved the issue. And that's pretty consistent with previous batches.

If you have the original IKEA controller, can you perform the recalibrate/reset procedure with it? I wonder if this is another edge case with the motors just very slightly out of sync and have drifted over the last year between each other just 1-2mm. The original IKEA controller might account for that between motors - the megadesk does not.

JanBe, 2020-10-19:
I am leaving it plugged in 24/7. 

I still have the original controller and will test out if recalibration fixes the issue.

gcormier, 2020-10-19:
Okay great, keep me posted!

gcormier, 2020-10-21:
Any luck?

closed 2020-11-03 by gcormier