gcormier/megadesk #32: Megadesk v4 no longer controls motors to adjust BEKANT height

mattpollard, 2020-11-08:
👋🏻 I have a Megadesk v4 that I bought on July 2, 2020. I've been using it happily in my BEKANT with two pre-programmed positions. The module's stopped working, and I'm hoping someone has an idea about how I can troubleshoot 🙏🏻

If I recall correctly, I was adjusting the desk from a standing height, down into my usual sitting position. For me, that's two short presses on the up button. For some reason, while the motors were working and the desktop was moving, I pressed either the up or the down button while the desk was moving between positions. The motors stopped and the Megadesk v4 module hasn't been able to move them since 😨 The module beeps for various actions (like the initialization handshake and if you try to store new positions), but the motors just never move. I _think_ you can hear the motors activating for just a split-second when you press the up or down button.

I appreciate the excellent README in this repository, and I've tried all six steps in the [Troubleshooting section](https://github.com/gcormier/megadesk#troubleshooting). Still nothing, though: the motors won't raise or lower the desktop. I've reinstalled the stock module and it works fine.

I'm happy to share a video or more detailed description of the module's behavior—but before I do that, I thought maybe someone might've seen something similar to this and have an idea offhand. (I didn't see anything similar in any existing issues, either.)

Thanks for any ideas 🙇🏻 (And thanks for Megadesk, @gcormier! It's such a nice little quality of life improvement. I'll buy another module in a heartbeat if I need to 🤘🏻)

gcormier, 2020-11-08:
Hi Matt!

Can I just confirm you did the recalibration procedure with the stock
controller and that was successful?

mattpollard, 2020-11-11:
I did! And I just did it again, and it didn't work. However, after that, I toggled to variant mode and everything's working as I'd expect! I don't know what happened, but I'm happy 😄 Thanks again for your work on Megadesk... happy that mine's functional again!

closed 2020-11-11 by mattpollard

gcormier, 2020-11-11:
You're the 2nd to have that issue, I wonder if there is something causing
variant mode to get reset. I'll put a note on the troubleshooting guide.

mattpollard, 2020-11-11:
Cool, thanks. I did reset the Megadesk module while troubleshooting, so at some point, _I_ probably cleared the variant configuration. During my first round of troubleshooting, though, I did try re-enabling the variant mode... but that didn't help. Pretty sure I always got the same beep pattern: many beeps, then two three-beep patterns (root, ascending, back to root).

From what I can tell, the only difference this time around was that I had the Megadesk module disconnected for a longer period of time. I noticed that when I disconnected the power strip from the wall (and left the BEKANT connected to the strip), the module had enough residual power to continue beeping for almost one minute 😆

Should this happen again, I'll try to get more detailed troubleshooting notes for you. I tried to recreate the problem by activating my presets and then pressing buttons again while the motors were in the midst of raising or lowering the desktop, but the module never "locked up" again 🤔