gcormier/megadesk #34: What can be used as keypads?

sichacvah, 2020-12-31:
What can be used as keypads?

Hello! First of all thanks for that project! I was buy old bekant table  without controller recently and want to use our megadesk for it, but i don't know where and what keypads i can buy for it? Could you provide any tips for me?

gcormier, 2020-12-31:
So the up/down buttons are merely SPST which connects the respective pins to ground over the ribbon cable.

I do have a breakout board design which you can see in the testing video

[image]

You could use the breakout pins to connect any buttons to them via soldering. Or, you could mount the 2 buttons as shown, and 3d print an enclosure.

As for something pre-made, off the shelf, I am not sure what options there are - this is why I designed megadesk to use the existing case as I didn't want to deal with designing a case and 3D printing it. 

You could use something like this

[image]

Connected to the pin headers of the breakout board. (Only 2 of the buttons would work though)

gcormier, 2020-12-31:
Oh they do sell a 2-button version as well.

[image]

You could even just use adhesive tape or glue to put this on the front of a piece of wood and hide the circuit board behind it.

closed 2021-01-11 by gcormier

codingforfun, 2022-01-19:
I used a physical 2 way momentary switch.
[image]