gcormier/megadesk #35: BOM Notes

dbrgn, 2021-01-15:
Hi, I plan to build a megadesk controller myself!

Is there anything special to consider when ordering the parts from the BOM? For example, is the capacitor ESR relevant somewhere, or can any regular ceramic caps be used?

What kind of flex connector is J2?

(If you have recommended MPNs, that would be helpful.)

gcormier, 2021-01-15:
Hmm yes it looks like I haven't exported the proper fields to ibom.html. I'll update those.

Here's the PN's I've been using

Manufacturer Part   Number | Manufacturer | Digi-Key Part Number | Reference Designator | Description
-- | -- | -- | -- | --
1N4148WS | ON Semiconductor | 1N4148WSFSCT-ND | D1, D2 | DIODE GEN PURP 75V 150MA SOD323F
ATTINY841-SSUR | Microchip Technology | ATTINY841-SSURCT-ND | U3 | IC MCU 8BIT 8KB FLASH 14SOIC
MCP2003BT-E/SN | Microchip Technology | MCP2003BT-E/SNCT-ND | U4 | IC TRANSCEIVER HALF 1/1 8SOIC
1-84981-0 | TE Connectivity AMP Connectors | A100308CT-ND | J2 | CONN FFC FPC TOP 10POS 1MM R/A
22288030 | Molex | WM50013-03-ND | J1 | CONN HEADER R/A 3POS 2.54MM
MC78M12CDTRKG | ON Semiconductor | MC78M12CDTRKGOSCT-ND | U1 | IC REG LINEAR 12V 500MA DPAK
MC78M05CDTRKG | ON Semiconductor | MC78M05CDTRKGOSCT-ND | U2 | IC REG LINEAR 5V 500MA DPAK
C1206C104K5RAC7800 | KEMET | 399-C1206C104K5RAC7800CT-ND | C3,C4,C5 | CAP CER 0.1UF 50V X7R 1206
CC1206KRX7R9BB334 | Yageo | 311-3403-1-ND |   | CAP CER 0.33UF 50V X7R 1206
PKMCS0909E4000-R1 | Murata Electronics | 490-9647-1-ND |   | AUDIO PIEZO TRANSDUCER 12.5V SMD

There is no special attention needed for any of the parts except voltage regulators that must have 35V rating. ESR/etc do not matter.

closed 2021-01-27 by gcormier

dbrgn, 2021-01-27:
Thanks a lot, @gcormier!