gcormier/megadesk #40: Desk stopped raising / lowering, even with new power supply -- T POWER power supply works, but only in 1 cm increments

TheChuckster, 2021-02-01:
Need troubleshooting help:

- Tried using connector-compatible 32V 2A printer power supply from T Power -- desk moves 1 cm then relay clicks, then requires power cycle again to move again

- Replaced with new ROL Ergo OEM power supply from IKEA, desk does not move in original mode, desk also doesn't move, but relay DOES making clicking sound in VARIANT mode

- Buttons seem to work

- Cleared EEPROM

- I do not have the ORIGINAL BEKANT control to recalibrate/reset the motors

Any ideas?

gcormier, 2021-02-01:
This is a tough one as I do suspect calibration would help.

Can you try to lower the desk all the way to the bottom position as much as possible? (Realizing this will be rather painful and require many power off/on cycles and 1cm moves), and see if it behaves any better?

I think the next step would either be obtaining a replacement controller, or getting the calibration routine in megadesk. The latter is still on my todo list and will take some time, and then would also require you to flash the new firmware.

What size desk do you have, large or small?

TheChuckster, 2021-02-06:
Do you have a spare original controller I can purchase? Do you also accept donations? -- really interested in seeing the calibration routine working ;-)

gcormier, 2021-02-08:
At this point in time I think the code/flash is nearly maxed with serial functions enabled, but it's still on my to-do list. Unfortunately as it's rare to recalibrate and the stock ones does this, it's not super high on the priority list, but I will try to rig something up to capture the packets.

I know I read somewhere else that a user had success talking to IKEA support on the phone and getting them to send them another controller when theirs broke - have you tried calling them?

TheChuckster, 2021-02-09:
I'll see what I can do. They are not responsive to phone calls. Will keep you posted once I manage to get my hands on one again and try recalibrating.

foosel, 2021-03-04:
FWIW, years ago I wrote them an email (or maybe a support ticket? Sorry, memory is hazy, this was early 2016) if I could get a second controller due to issues my first one and I got one sent pretty quickly.

orther, 2021-03-17:
I had the same issue with the desk only going up in ~1cm increments until I read online if I unplug the desk for a day or so it will start to worm again. I unplugged it for two days and when I plugged it back in it started working like usual. 

The problem is if you leave it plugged in to power the desk starts doing to 1cm steps again. To solve my issue I ordered the following power foot pedal switch to prevent electricity from going to the desk until I step on the pedal.

https://www.amazon.com/dp/B08BZX4XGY/ref=cm_sw_r_oth_apip_J6hbX3L434T2V

Not sure if this is helpful but figured I'd share in case it was.

gcormier, 2021-03-17:
Going to move this over to discussions. I do wonder if based on what orther said if it is a power supply issue - otherwise it could be something where the motors are refusing to move?

closed 2021-03-17 by gcormier