gcormier/megadesk #42: Had to cycle 2 times to alternate mode and back for it to work

alexbogun, 2021-02-14:
As the title says, did not work out of the box. Had to press up 16 times to get to alternate mode - still did not work. Pressed up 16 times again to get back to regular mode - worked like a charm.

Desk: Bekant small production 2020.

This is meant more like feedback than an issue, so you can close it straight away.

Great job anyway!

alexbogun, 2021-02-14:
Suggestion: maybe mention this as a workaround in your troubleshooting section?

gcormier, 2021-02-14:
I suspect that this is not actually a result of toggling back and forth, but rather in doing so, you did a reset and got a proper initialization - which is noted in the docs :) The handshake fails from time to time so perhaps you got unlucky the first time.

alexbogun, 2021-02-14:
I did power cycle it couple of times before, so not sure

closed 2021-03-02 by gcormier