gcormier/megadesk #43: Flashing troubles

dbrgn, 2021-02-16:
I finished building a v5 PCB 🙂🎉 

[image]

Unfortunately flashing the ATtiny doesn't seem to work:

```
$ avrdude -c usbasp -p t841 -U lfuse:w:0xe2:m

avrdude: error: program enable: target doesn't answer. 1
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
```

I used an USBasp programmer (with 5V output) and the current SVN version of avrdude (because the latest release doesn't support the t841) to flash the megadesk PCB. The PCB was only connected to the programmer, not to the desk.

I measured the voltage across the 5V and GND pins on the programming header, and it shows 4V. Without the megadesk PCB connected, the voltage provided by USBasp is 5V. Could be some kind of short, but the components don't get warm...

Unfortunately the schematics in KiCad look all garbled to me, I probably have the wrong symbols? (Incredible, that in 2021 this still doesn't work properly.) Should the 5V power supply from the PCB suffice, or is the 24V power input from the table required as well, for in-circuit programming?

gcormier, 2021-02-16:
For ICSP, all that is required is the 5V.

Is it the lighting, or is there a chunk missing from your attiny841?! :)

dbrgn, 2021-02-16:
Ooof, I think I got D1 soldered in the wrong way around. Those little bastards are really hard to read properly.

Do you have a PDF of the v5 schematics, so I can check what I could possibly have destroyed with that?

> Is it the lighting, or is there a chunk missing from your attiny841?! :)

Just the lighting, due to some flux residues 😂

gcormier, 2021-02-16:
Sure here you go
[image]

gcormier, 2021-02-16:
Which programmer are you using? At least you're on linux.

`avrdude` sure has given me some "fun" times.

gcormier, 2021-02-16:
Note for V5, if you are using an IDC connector and the typical AVR programmers, the notch on `v5` points up.

[image]

dbrgn, 2021-02-16:
I'm using an usbasp variant: https://www.fischl.de/usbasp/ I've used in the past for Arduinos and ATtinys and it worked fine so far.

Ok, so if I have D1 reversed (which I checked with a multimeter) then it connects 5V to 12V, which feeds backwards into the regulator. Probably nothing that should destory the regulator, but I'll double-check after fixing it.

> Note for V5, if you are using an IDC connector and the typical AVR programmers, the notch on v5 points up.

Thanks for the hint. I used jumper wires for now, since my programmer has a 10-pin connector 🙂 I think the reversed diode should be the reason, dropping the supply voltage below the 5V required to run the ATtiny.

I'll fix this another day. Thanks already for your quick help!

gcormier, 2021-02-17:
Great, let me know how it goes!

You can do all the programming, and you should hear the startup beeps with merely the 5v/programmer connected. No need to connect the desk to hear the beeps. Once you hit that point you should be good to switch over.

dbrgn, 2021-02-17:
> You can do all the programming, and you should hear the startup beeps with merely the 5v/programmer connected. No need to connect the desk to hear the beeps.

Ah, very helpful to know! Thanks!

dbrgn, 2021-02-20:
Hm, I fixed the diode, but as soon as I connect the 5V input to the ICSP header, it drops to 4V again.

Do you see potential for a component to get destroyed by the reversed D1? I'm a bit stuck with debugging. The diodes are still fine, according to my multimeter. I guess as a next step I'd probably desolder U2 to see if the problem still happens that way.

gcormier, 2021-02-21:
Do you have another attiny841?
Do you have a bench PSU capable of measuring current? Or can you hook up some power to it with a meter to measure the current? It should be under 30mA at all times.

closed 2021-03-02 by gcormier