gcormier/megadesk #51: 16 UPs does not activate Variant Mode

savenelson, 2021-03-28:
Hi,

I have the larger Bekant and have followed these steps:
1. w/ original controller,  calibrated the legs,
2. installed megadesk controller, powered on (4 beeps)
3. 16 UPs and no sounds - tried 20 UPs (no beeps though).
4. Unplugged controller, unplugged desk, plugged in desk, plugged in the controller (4 beeps).
5. 16 UPs (no beeps)
6. unplugged desk, held DOWN while plugging in, to reset EEPROM (no beeps)
7. unplugged desk and plugged back in (4 beeps)

I'm not sure what step to try next - I've done the above cycle a few times and cannot activate variant mode to account for my larger Bekant.

Thank!

gcormier, 2021-03-28:
Hello!

So I checked your order and it was the last one before the new firmware... it is possible you may have gotten the new firmware, which auto-adjusts for the variant.

Can I get the build date on your unit? Should be on a sticker on the back side of your controller like those found in https://github.com/gcormier/megadesk/issues/46

savenelson, 2021-03-28:
Hi Greg,

It's 2020-37

gcormier, 2021-03-29:
Hmm, so it should pickup the variant automatically. Note that the
power-cycle, you need to wait a good 15-30 seconds, as the power supply has
some huge reserves that will power the tiny controller.

savenelson, 2021-04-09:
Fixed!

closed 2021-04-09 by savenelson