gcormier/megadesk #52: Serial Control - Question

jkms, 2021-03-29:
It looks like serial control has been added.  I just have a couple of questions.

- Is this just a software change?  Or is a hardware revision pending for the device on Tindie?
- There doesn't seem to be any documentation yet.  Are the pins labeled on the board?
- I think you said the default firmware will not enable this?  Do I have to compile the firmware with option enabled?

gcormier, 2021-03-29:
Yes, just a software change - you will need to compile and upload the firmware as outlined in the DIY readme.

I guess I do need to beef up the documentation a bit - you'll be using 3 pins on the programming header for GND/TX/RX.

jkms, 2021-03-29:
Great!  I'm going to order a board right now!  Very excited.

closed 2021-04-24 by gcormier