gcormier/megadesk #58: How do I use the serail commando?

rhannink, 2021-05-19:
Hello,

First off all, great product, works like a charm, but I would like to connect it to my domotic system (HA) and Google system.
I recompiled the code with the #define SERIALCOMMS.

I flashed the firmware with a pocket AVR programmer and this all works fine. 

When I connect an FDTI232 to the pins MISO and SCK I see the following commands in my terminal when I press the up or down button.

```14:51:02.422 -> <+  <+  <+  <+  <+  <+  <+  <+  <=z <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=⸮ <+ 	 <=⸮ <- 	 <-  <-  <-  <-  <-  <-  <-  <-  <-  <-  <-  <-  <=⸮ <-  <=⸮ <-  <=⸮ <-  <={ <-  <=f <-  <=Z <- 	 <=Q <-  <=N <+  <+  <+  <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=⸮ <+ 	 <=⸮ <- 	 <-  <-  <-  <-  <=⸮ <-  <=⸮ <-  <=o <-  <=W <-  <=B <-  <=6 <- 	 <=- <-  <=* <+  <+  <+  <+  <+  <+  <+  <+  <+  <+  <=⸮ <+  <= <+  <=) <+  <=> <+  <=P <+  <=\ <+ 	 <=e <- 	 <-  <-  <-  <-  <-  <=⸮ <-  <=⸮ <-  <=⸮ <-  <=⸮ <-  <=⸮ <-  <=⸮ <- 	 <=⸮ <-  <={ <+ 	 <+  <+  <+  <=⸮ <+  <=⸮ <+  <=⸮ <+  <=```

So that looks ok, but how can I sent a command via a serial terminal program? I tried to sent <- and <+ with or without returns, but nothing happens. What am I missing. Do i need to sent more bytes ? Can I use a serial terminal command line at all to sent commands?

Eventually I would like to use an ESP(8266 or 32) to control the megadesk. Maybe someone has some example code on what command/bytes exactly to sent over the serial line?

Thanks in advance
Remco

tagno25, 2021-05-22:
Here is where I described the command structure
https://github.com/gcormier/megadesk/issues/12#issuecomment-766170915

The commands are mostly binary, with ASCII command indicators.

rhannink, 2021-05-22:
Thanks,

Ik will look into that...

rhannink, 2021-12-06:
Hello,

Finally found time to implement the serial commands. I flashes the latest firmware with the serial commands enabled and connected a esp32 to the serial port. I loodde the following ESPHOME code onto the ESP and not I can control my Ikea desk from Home Assistant and through Google Home (**Ok Google activate standing/sitting/middle**)

`esphome:
  name: desk
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "xxxxxxxxxxx"
  password: "xxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Desk Fallback Hotspot"
    password: "xxxxxxxxxx"

captive_portal:

uart:
  baud_rate: 115200
  tx_pin: D0

switch:
  - platform: uart
    name: "Desk up"
    data: '<L0,4.'
  - platform: uart
    name: "Desk down"
    data: '<L0,3.'
  - platform: uart
    name: "Desk middle"
    data: '<L0,5.'`

Works great, also the possibility to recalibrate and set the min and max heigth is a welcome addition.

Gr. Remco

gcormier, 2021-12-06:
Wow, that's awesome that it was so easy with esphome! Nice work!

rhannink, 2021-12-06:
Only to bad, the power regulators of the megadesk can only deliver 5V/500mA. Now I need an additional powersupply to feed the ESP. Maybe in a new version you can upgrade the regulators to 1A or 1.5A types (they have same package I think). And maybe add an extra sideways connector with RX,TX,5V and GND. The current header cannot have a paring connector due to the height of the housing.

Keep up your great work...

Remco

gcormier, 2021-12-06:
The issue is not the part itself, but the thermal limitations of a linear regulator in a small space.

https://www.gadgetronicx.com/calculators/linear-regulator-power-dissipation-calculator/

It is best to tap the 24V(30V) independently to power the secondary MCU with an independent regulator, DC BUCK or linear.

rhannink, 2021-12-07:
Maybe it helps if you cascade the 7812 and the 7805 to lower the voltagedrop, but the 7812 then needs to deliver more power, still a lot of heat dissipation.
I will use a buck converter to bring down the 24V to 5V.

Thanks.

gcormier, 2021-12-07:
At the end of the day there is still around 7-13W of power to dissipate. A buck would be best for sure :)

benfysh, 2022-03-13:
@rhannink have you documented this in more detail anywhere? Thanks

rhannink, 2022-03-16:
Hello @benfysh,

I did not find the time to document this project on GitHub. What I did was:

- Buy Megadesk controller on Tindie. https://www.tindie.com/products/gcormier/megadesk/
- Flash the controller with the latest firmware with serial commands enabled. You need the [Sparkfun AVR programmer](https://www.sparkfun.com/products/9825) for this
- Flash a Wemos D1 mini (or any other ESP) with the [esphome](https://www.esphome.io) software.
- Connect de D0 pin from the ESP8266 (TX1) to the RX pin of the megadesk (PIN 4 MOSI)
- Power the ESP8266 from the Ikea Desk by using a 24-5V Buck converter
- Configure the ESP8266 in Home Assistant with the following settings:

``` 
 name: desk
  platform: ESP8266
  board: d1_mini

logger:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "xxxxxxxxxxx"
  password: "xxxxxxxxx"

  ap:
    ssid: "Desk Fallback Hotspot"
    password: "xxxxxxxxxx"

captive_portal:

uart:
  baud_rate: 115200
  tx_pin: D0

switch:
  - platform: uart
    name: "Desk up"
    data: '<L0,4.'
  - platform: uart
    name: "Desk down"
    data: '<L0,3.'
  - platform: uart
    name: "Desk middle"
    data: '<L0,5.'
```
Now you have 3 switches available in HA for 3 different presets.

tobru, 2022-03-16:
I wrote a blog post about it, maybe it helps: https://tobru.ch/ikea-bekant-table-control-over-mqtt-with-megadesk-and-esp32/

benfysh, 2022-03-16:
@rhannink & @tobru thanks both looks like I have a few more things to buy :) Megadesk is already working well for me I just like the idea of automating it and running automation when it is set to different heights.

rhannink, 2022-03-16:
@benfysh , indeed it is very cool, once it is hooked up to HA you can also control it with Google Home. "Ok Google I want to standup", "Ok Google, I want to sit down".

gcormier, 2022-03-17:
Gonna add a todo to bring out some info in the docs so people don't have to dig in issues for the commands.

gcormier, 2022-03-20:
Added serial/esphome documentation to bottom of readme.

closed 2022-03-20 by gcormier