gcormier/megadesk #64: Default Firmware

gcormier, 2021-05-27:
Moving forward I'd like the standard build that I ship to include the min/max height setting as this would be great.

1. From all the hard work @philwmcdonald  did, could/should move BOTHBUTTONS to parameter territory? It would default to disabled, but not require reflashing if someone wanted to enable it.

2. I think SERIAL would be left out, as anyone who's getting into hacking up serial functionality should probably have the skillset to compile and reflash?

3. Pushing both buttons and holding is still up for grabs for a function if we'd like to quickly access.

philwmcdonald, 2021-05-27:
I agree on min/max being default feature - I do think it needs better audio feedback so that it's clear whether you just set or cleared a limit. Made this mistake the other day.

Serial is definitely for the sophisticated user, because there's much additional work to interface/control it. (I haven't made any use of it yet).

We don't want users to accidentally change modes that drastically change the behaviour. If bothbuttons toggle happens unexpectedly that's confusing.

I'd like to see toggling single/both button mode, or running the recalibration done only during power-on with a button (up, down, both) hold. but recognize that's a difficult decision. I'm not sure the existing button-test provides much value - With bothbutton on for instance double clicking either button provides audible feedback even if the motors don't operate. 

Of course, diverging behaviour from the existing population of megadesks creates two versions to document/support, most existing will likely never be upgraded with the latest software.

tagno25, 2021-05-27:
I agree on the serial feature requiring the user to compile and flash the firmware, at least for existing boards.

If you make a future board revision and there are unused pins on the ATtiny, then maybe serial could be compiled by default and enabled with a jumper or solder bridge.  Or if you switch to a ATtiny1624, then keep requiring the user to compile and flash the firmware, since flashing is even easier with the tinyAVR 2 family.

gcormier, 2021-05-27:
I would be up for modifying the startup routine. Holding DOWN during power on could reset EEPROM, and then enter the button test mode afterwards. This has little effect on existing userbase as DOWN is expected to wipe EEPROM anyways. The button test is rarely used but helps me ensure the users have seated the ribbon cable fully.

This would leave holding UP for a function during startup. Perhaps the both button toggle? I also imagine that is a rarely changed parameter, you either want it or you don't.

TinyAVR-2 series is great, but chipageddon makes it unobtanium. And the PlatformIO support is a bit lacking at the moment. I wouldn't be surprised if getting these are impossible in the next 12 months.  The attiny841 is now out of stock at digikey in SOIC, and MCP2003B is fairly low, although they must have gotten a small shipment as they are showing 2000 units. Sad time for hardware design!

philwmcdonald, 2021-05-28:
PR #66 is up now with these software ideas implemented. 
It's bigger, but it has more features. still fits in 8kb with serial.

gcormier, 2021-05-28:
Regression at some point, was not #66 but moving UP no longer coasts/decelerates. Will investigate this weekend.

gcormier, 2021-05-28:
d944d1f is sus

philwmcdonald, 2021-05-28:
It was me.
#61 made lastState local.
which breaks the state machine you have for deacceleration.

gcormier, 2021-07-21:
I think I'm ready to merge `dev` into `master` after soaking the latest build for some time. Have either of you hit any issues?

The readme in the `dev` branch has a new 'megaupdate' section with the changes from the last few months, let me know if I've missed anything!

philwmcdonald, 2021-07-22:
I'll take a look in the next few days.
5 push slot currently has reset functionality, that should be moved to another slot before merging to master.

philwmcdonald, 2021-07-22:
I haven't had the programmer or serial connected to my desk in a while so I'm not exercising it as much. It's been up and running 3 weeks without any obvious sign of trouble.

closed 2021-10-31 by gcormier