gcormier/megadesk #65: Change ICSP header

tagno25, 2021-05-27:
I would recommend changing out the straight ICSP for a right angle header.

If the header is a right angle then it is possible to cut a hole in the back of the existing case, which allows a cable to easily connected and disconnected without disassembly. It didn't seem to interfere with the case if you don't cut a hole.

gcormier, 2021-05-28:
Is there an action needed here? Changing BOM/notes? Is the current layout OK?

tagno25, 2021-05-28:
I think the current layout is OK.  

I don't recall if the boards came with a header already populated at this point.  If they don't then just changing the DigiKey link in the DIY.md file would probably be sufficient.

https://www.digikey.ca/en/products/detail/amphenol-icc-fci/68459-406HLF/5359607

philwmcdonald, 2021-05-28:
> I would recommend changing out the straight ICSP for a right angle header.
>
> If the header is a right angle then it is possible to cut a hole in the back of the existing case, which allows a cable to [...]

I did exactly this with the rev4 board. The pins extend out the back of the case. I can keep them covered with a socket or plug in serial or the avr programmer. I didn't have the part you reference, instead I bodged/soldered right-angle and straight header pins.

philwmcdonald, 2021-05-28:
[image]

gcormier, 2021-05-31:
a287e9a

closed 2021-05-31 by gcormier