gcormier/megadesk #68: potential to move beyond DANGER_MAX_HEIGHT/DANGER_MIN_HEIGHT?

philwmcdonald, 2021-05-29:
with the recalibration procedure the safety margin was removed. I'm wondering if this is still safe:
```
if (targetHeight > currentHeight && abs(targetHeight - currentHeight) > HYSTERESIS && currentHeight < maxHeight)
    motorUp(true);
  else if (targetHeight < currentHeight && abs(targetHeight - currentHeight) > HYSTERESIS && currentHeight > minHeight)
    motorDown(true);
```
or if that should be changed to:
```
if (targetHeight > currentHeight && abs(targetHeight - currentHeight) > HYSTERESIS && targetHeight < maxHeight)
    motorUp(true);
  else if (targetHeight < currentHeight && abs(targetHeight - currentHeight) > HYSTERESIS && targetHeight > minHeight)
    motorDown(true);
```
This wouldn't effect recalibration, as that this snippet isn't executed during recal.

gcormier, 2021-05-29:
Hmm good question. I think we should be OK as we still have `HYSTERISIS` which is set at 137 that is added/removed from the boundaries. It doesn't necessarily match the coast length for the deceleration which I haven't been able to measure.

I can plug in the logic analyzer again and see what the reported motor heights are for both controllers at their limits to see if we need to adjust.

philwmcdonald, 2021-06-04:
I monitored serial which helpfully provides telemetry during the coast stage (after memoryMove is off).
I measure movement of 120 & 123 going upwards and 111 & 111 going down. Both are less than hysteresis of 137. but typically each time through the loop we see a movement of 24 or even 27, currentHeight may already be 20+ beyond min/max.

I set max and a slot at the same height. then tried both memory recall and up-button-hold to see if I could exceed the max I set.
memory recall never exceeded the max, but I could with the up-button. and exceeded the max by the coast distance I measured above. 

I have a fix and it isn't just replacing currentHeight with targetHeight as above. That has other unintended effects.

closed 2021-06-04 by philwmcdonald