gcormier/megadesk #77: First DOWN after powerup not working?

gcormier, 2021-07-26:
Flashing the current megaupdate builds... after plugging in a board, pushing DOWN button has no effect. Pushing UP button moves the desk, and subsequent DOWN button works without issues.

philwmcdonald, 2021-07-29:
I couldn't repro this. Anything special about your setup? 
Down button is held for >0.4s right?
I'm guessing feedback and bothbuttons options are off.

gcormier, 2021-07-29:
Hmm odd. I can't repro it now on my main unit. But it was the case for 50 boards.

Basically, it's flashed on a desk with no connection to a desk, and then powered down. Rinse and repeat. Then I bring a batch upstairs and test each one.

I wonder if it has to do with the auto detection of the IDLE value the first time it is up and running?

philwmcdonald, 2021-07-29:
There's no state info that survives a reset apart from the contents of
eeprom. None of the eeprom slots contain anything that should impact this
either.
However your desk keeps state: the encoder values on each leg/motor...
Resetting the freshly flashed/installed controller after it can't descend
does that work? How about resetting after it *can* descend?

Phil
613-600-3105

gcormier, 2021-08-04:
This is on my todo list for this weekend...

gcormier, 2021-08-14:
Okay, I swear I'm not crazy, but I can't repro this either right now :) 

Next batch of assembly, I'm going to pay a bit more attention and check it out.

philwmcdonald, 2021-08-19:
It occurs to me that if the motors don't report idle then movement isn't possible. That should apply to any movement, so that's not a likely cause here. enabling SERIAL_IDLE allows that to be triaged/eliminated as a cause.
Also I enabled SERIAL_IDLE and notice that the idle value isn't always consistent:
sometimes the motors report:
>i0,0
sometimes:
>i37,37

philwmcdonald, 2021-08-23:
I'm seeing if motor a reports 37, and b reports 0 It gets stuck in STOPPING4 state, and never goes idle.
For me that's the state where manual up/down buttons don't work.
What confused me was that memory recall shouldn't work either, I believe the difference is that the memory-recall beeps delay the regularly scheduled linBurst() long enough that both motors report the same value, transitioning to OFF state and allows subsequent movement.

gcormier, 2021-09-12:
Confirmed fixed! :)

closed 2021-09-12 by gcormier