gcormier/megadesk #78: Enhancement request: add breakouts for button connectors

filoxo, 2021-08-12:
Hi! I just ordered some of these and I was unfortunate enough to cut through the ribbon cable (my fault, of course).

Looking at the original controller, it appears to have breakouts that map to the pads that the ribbon cable connects the tactile buttons to. The Megadesk PCB does not. Below is a picture of the original controller (top) that shows what I'm referring to.

[image]

I think it'd be a great feature to break those out for additional tinkering; either as through holes (underneath like the original) or as smd pads. Either implementation can be used to solder headers or wires directly. Then poor souls like me to more easily adapt buttons like those recommended in #34 without the need of yet another daughterboard, as well as let people bring their own buttons such as mechanical keyboard switches.

I don't think there's anything else I can contribute to this idea so feel free to close at your discretion. Thanks you!

gcormier, 2021-08-13:
I can probably add some underneath the FPC connection.

If you send me your order #, I do have some spare breakout boards, I could send you one if that helps as well as a spare ribbon cable.

codingforfun, 2022-01-19:
I used this from Amazon (you probably also find them elsewhere). They have the same spec as the original ribbon cable. I just soldered a short piece to wire and put it together with megadesk in a custom case. The short extra piece of ribbon wire was no  problem to me since I hide the board under the desk separate from the physical switch I added. So need wire anyway. 
https://www.amazon.de/gp/product/B016DDJAGM/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1

+1 for the idea of button breakouts. Just wanted to share my solution with your for the meantime.

gcormier, 2022-01-20:
Nice!

I could try to make a PCB the same size as the front of the IKEA panel. It could side inside there with double-sided adhesive perhaps. It would be a bit proud of the case but still functional.

gcormier, 2022-03-20:
https://github.com/gcormier/megadesk/releases/tag/pcb-v8-C668117

This release (v8) adds 2 SMD pads for up/down. Ground can be grabbed from programming header.

[image]

News coming soon about breakout boards!

closed 2022-03-20 by gcormier