gcormier/megadesk #81: I'm tall. When the table is goes to the upper limit no presets seem to respond.

frankhommers, 2021-09-19:
See subject ;-)

When I unplug the the power cord it and re-plug it it starts to work again.

Also I found a workaround by just stopping underneath the upper limit.

Is this a known bug?

gcormier, 2021-09-19:
Okay so just to be clear, you have a memory stored which is as high as the table will go? And once you recall this memory it will no longer move?

frankhommers, 2021-09-20:
No when I move the table to the upper limit, and I recall a preset (of course below that level)  it does not work.

philwmcdonald, 2021-09-30:
I tried to reproduce on most recent s/w on dev branch and couldn't.
Lots of changes were made w.r.t limits and some fixes to idle-states which previously required power-cycle to recover. I'm not convinced these would fix what you're seeing, but who knows. 
What version of the software do you have?

frankhommers, 2021-09-30:
I'll post back to you when I flashed it. I am not sure atm. I am waiting for my AVR programmer to arrive.

frankhommers, 2021-10-05:
I still have this with v2021.09-serial.hex. It's not a really big problem because I use a preset just below the top position.

gcormier, 2021-10-31:
Closing since we can't repro not does it seem like more info is coming.

closed 2021-10-31 by gcormier