gcormier/megadesk #83: Can I power a Wemos D1 Mini v3 from 5V + GND ports?

frankhommers, 2021-09-19:
Would it be possible to power a [Wemos D1 Mini v3](https://www.bol.com/nl/nl/p/wemos-d1-mini-v3-0-0-wifi-internet-of-things-development-board-gebaseerd-esp8266-4mb-micropython-nodemcu-arduino-compatibel/9200000116017964/) from the 5V + GND pins? Or would it be better to use a separate power supply for that?

gcormier, 2021-09-19:
I would ***definitely*** recommend a separate power supply, as the regulators on the megadesk are already doing a lot of work to drop the 24V down to the needed 12V and 5V. While the regulators still have capacity in terms of milliamperes to provide, they do not have much thermal headroom.

You can read more
https://www.sparkfun.com/tutorials/217
https://electronics.stackexchange.com/questions/431361/lm317-calculate-dissipation-due-to-voltage-drop

You could tap the 24V incoming power (red wire) and ground (black) to run to a secondary power supply to power your D1 mini.

frankhommers, 2021-09-20:
Thanks I will do that.

closed 2021-09-20 by frankhommers