gcormier/megadesk #84: Board seems to keep restarting

JorritPosthuma, 2021-09-22:
Hi! First of all, thank you so much for the project. It's been working like a charm as a smarter replacement for the IKEA version. I also wanted to make it even smarter by connecting it to an ESP32 with Apple Homekit :), but I've ran in to some problems:

- When I connect the board to the SparkFun AVR programmer, it keep repeating the "BOOT" beeps every 2 seconds. However, I was able to flash the firmware with the Serial feature enabled
- When I connect it to the ESP32 it does the same, and I keep receiving `*E0,2` over serial (also repeating every 2 seconds), and it's not responding to any commands
- When it gets its power from the desk though, everything (still) works as expected

Do you have any idea what I could have done wrong? It might be my soldering skills as I'm still relatively new to the hobby of electronics :).

gcormier, 2021-09-22:
It is unable to talk to the motors, so it resets itself. This is to be expected when programming and not attached to the desk. There is an option that will disable this reset.

For the ESP32, this is when everything is connected?

JorritPosthuma, 2021-09-22:
Thanks for the quick reply! That makes sense :). I had everything connected with the ESP32 powered by the 5V of the pinouts of the board, but some components got hot pretty quickly, so that's probably not the right way to connect everything up 😅. 

A little outside of the scope of my original question, but could you suggest how you would power the ESP32? I imagined that with 2 different power supplies (the desk for the megadesk board, and a separate 5V PS for the ESP32) that would only lead to trouble? Or is that ok if you connect the grounds? (Sorry for the basic electronics questions 😅)

JorritPosthuma, 2021-09-22:
Thanks! I’ll do some more reading ☺️. Have a nice day!

closed 2021-09-22 by JorritPosthuma