gcormier/megadesk #88: Megadesk keeps restarting

HerrLevin, 2021-12-13:
I've purchased one of your boards on tindie but I can't get it to work.

As soon as i connect it to the desk, it keeps restarting (it always repeats the three-tone-chime). When I press the down key, I can get into the key-testing-mode. 

I've already tried all troubleshooting-steps without any success. There also seems to be no voltage drop once I plug it in. Supply voltage keeps a stable 30V and the data line is around 12V once the board is plugged in.

Is there something else I can do?

gcormier, 2021-12-13:
The beeping-loop means it's having trouble on LIN talking to the motors.

Can you check your cable and the 3 pins for continuity? And check the black end of the connector that all 3 metal crimps are securely pushed and clicked in place?

gcormier, 2021-12-16:
Closing issue due to inactivity, if you still need help please feel free to reopen.

closed 2021-12-16 by gcormier

HerrLevin, 2021-12-18:
The pins have perfect connection but I've noticed that the order of the single wires changes from red->yellow->black on the board-side to red->black->yellow on the desk-side. Is this intentional?

gcormier, 2021-12-18:
Hmm that is odd. Black is ground and should be the center pin on both. You
can reorder the board side by removing and swapping the pins

https://youtu.be/-InoAbkNVdQ

HerrLevin, 2021-12-18:
Yes, that fixed it! Thank you very much!

gcormier, 2021-12-18:
Awesome! Now I will need to go check all my cables before sending...