gcormier/megadesk #89: Megadesk not controlling Bekant

ddemakes, 2022-01-01:
Happy New Year!

I had been using Megadesk V5 controller with Bekant with the date 2019-48 rec:C since Nov 2020.  About a month after install the presets failed.  The controller still allowed up and down control, but wouldn’t accept new presets.  

After nearly a year of not having to figure out a solution to the presets not being accepted, I finally had time today to try to figure it out, but the controller still wouldn’t accept them.  I cycled power both ways as described in troubleshooting, but that didn’t work.  

I tried the button connectivity test, and do hear the up chime and down chime when I press up and down.  I then cleared all memory, still not controlling the table.  

I can’t find the OEM board, so I haven’t been able to recalibrate/reset the motors.  I also don’t have the first clue about how to flash the board to add the factory recalibration routine from the July update, so my desk is stuck at a terrible height between sitting and standing.  What can I do?

Best regards,
Darrell

gcormier, 2022-01-01:
It's odd it went into a state where the memory positions did not work but manual positions continued to work.

If you wiped the memory, did you then toggle variant mode with 16x button pushes?

Flashing the board is not too difficult if you have some basic electronics, or the will to learn them. No soldering required.

ddemakes, 2022-01-01:
So, I toggled the variant and it works now.  Which is interesting, because I don't remember toggling the variant when I first set the desk up.  Covid memory???

But it works.  Cool.  Thanks Greg.  

About flashing the board.  When you say "basic electronics" what are we talking about?  How would I connect to the board?  Is there a USB to ??? connector I can purchase online?  Could you point me to a good primer online?

Thanks for your help.

Best,
Darrell

gcormier, 2022-01-02:
Awesome glad that fixed it up :)

Basically you'll need an AVR programmer, which can be bought from various sources (incuding amazon) for cheap. 

https://github.com/gcormier/megadesk/blob/master/DIY.md#programming

Cable an 6 pin header.

The firmware HEX file is available, so you don't need to do any compiling or installing of IDE's/etc. You can simply download `avrdude` and tell it to flash the MCU with the HEX file.

gcormier, 2022-01-02:
I'll point you to this discussion that has a bit more info and we can continue the conversation there.

https://github.com/gcormier/megadesk/discussions/49

closed 2022-01-02 by gcormier

ddemakes, 2022-01-02:
Would this work?
https://www.amazon.com/Standard-Adapter-Programmer-ATMEGA8-ATMEGA128/dp/B07KKGSSBN/ref=sr_1_4?crid=2EHBW739C7ON3&keywords=avr+programmer&qid=1641137242&sprefix=avr+programmer%2Caps%2C81&sr=8-4

gcormier, 2022-01-02:
Yes that would work. You need a 6pin header to connect it to the circuit board though (2x3, 2.54mm pitch), eg https://www.amazon.com/Break-away-Headers-2-54mm-Length-Centered/dp/B07842GC5F