gcormier/megadesk #91: Audio feedback when not doing anything?

davidstosik, 2022-02-04:
Hello there, just received and installed my Megadesk, and I'm super happy with it!

I enabled the audio feedback (memory position 17), and I like getting feedback on my button presses (the increasing pitch is genius!).

I noticed however that every now and then, the Megadesk will output a single beep, although I haven't touched it in hours. I imagine it's doing something like waking up and checking the desk position, but I was wondering if there would be a way to turn off that single beep while keeping the button press and result audio feedback?

Thanks a lot for working on this, it's great!

gcormier, 2022-02-04:
Hey David - I'll enable my audio feedback and see if I can repro this issue.

davidstosik, 2022-02-09:
Yup, that's correct. :)

gcormier, 2022-02-14:
I think it might be one of the audio feedback sections in linBurst(). I'll try to comment that out and see if that fixes things up.

gcormier, 2022-02-14:
Ok flashed, I'll see how it goes :)

If it does resolve the issue you will need a programmer to update your firmware. More info : https://github.com/gcormier/megadesk/blob/master/DIY.md#programming

davidstosik, 2022-02-15:
Yeah I thought that might be feedback in `linBurst()` too.
I probably have a programmer somewhere deep in my _stuff_. If I solder male pins, will it still fit in the case?

gcormier, 2022-02-15:
Yes it will still fit. Alternatively you can just press and hold a 6pin header in the through holes while programming. It's how I program the units I build.

gcormier, 2022-02-23:
I think this commit resolved it. Have you been able to flash yours to confirm?

davidstosik, 2022-02-23:
I haven’t yet. I need to find some free time to
- solder the header
- find my programmer
- setup platformio
- compile and flash

davidstosik, 2022-03-06:
An update: I couldn't find an AVR programmer in my stuff, so I had to figure out how to use an Arduino Uno as programmer:
- [Arduino as ISP on PlatformIO's documentation](https://docs.platformio.org/en/stable/platforms/atmelavr.html#:~:text=U%20flash%3Aw%3A%24SOURCE%3Ai-,Arduino%20as%20ISP,-%5Benv%3Aprogram_via_ArduinoISP%5D%0Aplatform%20%3D%20atmelavr)
- [Arduino as ISP and Arduino Bootloaders on Arduino's docs](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP)

It seems I managed to flash the board! 🎉 

Now I'll have to wait a few days and confirm whether I hear unexpected blips from Megadesk. 🤞🏻

gcormier, 2022-03-17:
Latest release is out and fixes this.

closed 2022-03-17 by gcormier

davidstosik, 2022-03-23:
Oh by the way, I think I haven't hear a blip since I flashed the board. 👍🏻 
Thank you!