gcormier/megadesk #98: Problems using with esphome

JonathanTreffler, 2022-03-25:
I bought a prebuilt Megadesk module, connected it to a ESP32 and flashed that using esphome and the provided yaml file.

I get the following logs:
```
INFO Reading configuration /config/esphome/megadesk.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing megadesk (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/megadesk/.pioenvs/megadesk/src/main.cpp.o
Linking /data/megadesk/.pioenvs/megadesk/firmware.elf
RAM:   [=         ]  12.8% (used 42020 bytes from 327680 bytes)
Flash: [=====     ]  50.5% (used 926694 bytes from 1835008 bytes)
Building /data/megadesk/.pioenvs/megadesk/firmware.bin
esp32_create_combined_bin(["/data/megadesk/.pioenvs/megadesk/firmware.bin"], ["/data/megadesk/.pioenvs/megadesk/firmware.elf"])
Wrote 0xf2450 bytes to file /data/megadesk/.pioenvs/megadesk/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 8.85 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of megadesk.iot
INFO  -> [redacted]
INFO Uploading /data/megadesk/.pioenvs/megadesk/firmware.bin (926800 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from megadesk.iot using esphome API
INFO Successfully connected to megadesk.iot
[22:07:48][I][app:102]: ESPHome version 2022.3.0 compiled on Mar 25 2022, 22:07:16
[22:07:48][C][wifi:491]: WiFi:
[22:07:48][C][wifi:353]:   Local MAC: [redacted]
[22:07:48][C][wifi:354]:   SSID: [redacted]
[22:07:48][C][wifi:355]:   IP Address: [redacted]
[22:07:48][C][wifi:357]:   BSSID: [redacted]
[22:07:48][C][wifi:358]:   Hostname: 'megadesk'
[22:07:48][C][wifi:360]:   Signal strength: -47 dB ▂▄▆█
[22:07:48][C][wifi:364]:   Channel: 6
[22:07:48][C][logger:233]: Logger:
[22:07:48][C][logger:234]:   Level: DEBUG
[22:07:48][C][logger:235]:   Log Baud Rate: 0
[22:07:48][C][logger:236]:   Hardware UART: UART0
[22:07:48][C][uart.arduino_esp32:107]: UART Bus:
[22:07:48][C][uart.arduino_esp32:108]:   TX Pin: GPIO25
[22:07:48][C][uart.arduino_esp32:109]:   RX Pin: GPIO33
[22:07:48][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[22:07:48][C][uart.arduino_esp32:113]:   Baud Rate: 115200 baud
[22:07:48][C][uart.arduino_esp32:114]:   Data Bits: 8
[22:07:48][C][uart.arduino_esp32:115]:   Parity: NONE
[22:07:48][C][uart.arduino_esp32:116]:   Stop bits: 1
[22:07:48][C][template.number:050]: Template Number 'Megadesk Height (inches)'
[22:07:48][C][template.number:050]:   Unit of Measurement: 'inches'
[22:07:48][C][template.number:051]:   Optimistic: NO
[22:07:48][C][template.number:052]:   Update Interval: never
[22:07:48][C][template.number:050]: Template Number 'Megadesk Height (cm)'
[22:07:48][C][template.number:050]:   Unit of Measurement: 'cm'
[22:07:48][C][template.number:051]:   Optimistic: NO
[22:07:48][C][template.number:052]:   Update Interval: never
[22:07:48][D][uart_debug:158]: >>> "<R0.12."
[22:07:48][C][template.number:050]: Template Number 'Megadesk Height (raw)'
[... 206 lines cut ...]
      then:
        - uart.write: "<L0,18."

interval:
  - interval: 300s
    then:
      - uart.write: "<C0.0."
```

I tried inverting the UART RX and TX logic levels:
```
[22:37:45][D][button:013]: 'Desk Position 2' Pressed.
[22:37:45][D][uart_debug:158]: >>> "<L0,2."
[22:37:54][D][uart_debug:158]: <<< "\x81\xD0\x03\x00\x00\x00\x00\x00\x00\x18\xFE\x01\x80\x10\x00\x00\x00\x00\x00\x02\x00\x98\x00\xC0\xE0\xE0\xE0\xF0\x00D\x00\xC0\x00\x00\x00\x98\x00\xFA\x00\x00\xFC\x00\xFE\x00\xF0\x00\xF0\x00\xD8\x00\xE4\x00\xF6\x00\xF4\x00\xF8\x00\x02\xE0\x00\x00\xFE\x00\xFC\x00\xFC\x00\xFE\x00\xFC\xE0\x00\xE0\xFE\xE0\x00\xFE\x01\xFC\x00\xFE\x00\xFE\x00\xFC\x00\xFC\x00\xFA\x00\xF8\x00\xFE\xFC\xE0\x00\xF8\x00\xFE\x00\xFE\x00\xFE\x00\xFF|\xFF\x00\xFF\x00\xFF\x00\xFF\x00\xFE\x00\xFF\x00\xFF\x00
[22:37:55][E][uart:015]: Reading from UART timed out at byte 0!
[22:37:55][D][uart_debug:158]: <<< "\xFF\x00\xFF\x00\xFF\x00\xFE\x00\xFF\x00\xFE\x00\xFE\x00\xFF\xFF`\x0E\xFF\x00\xFF\x00\xFE\x00\x00\x0F\x00\xFF\x00\xFF\x00\xFE\x00\xFFp\xFF\x00\xFE\x00\xFE\x00\xFCo\x01\x00\xB8\xFF"
[22:37:55][E][uart:015]: Reading from UART timed out at byte 0!
[22:37:55][D][uart_debug:158]: <<< "\x00\xFE\x00\x00>\xF6\x00\xFE\x00\xFE\x00\xFE?\x00\xFC\xFE\x00\xFE\x00\xF6\x03\xF6\x05\x80\xFE\xC0\x00 \x06\x00\xF0\xF0\x06"
[22:38:01][E][uart:015]: Reading from UART timed out at byte 0!
[22:38:01][D][uart_debug:158]: <<< "b\x00\x00\xA0x\xDF\xFF\xC7\xFF\x1FD\x80\n\xDD\xFF\xFF\xF7\x00\x87\x00\x00\x00\x00\x00\x00\x00\x00\x02\x0F\xE0\x00\x00\x00\x00\xE7\xFF\t\x01\x00\x18\x00\x00 \x00\xE0\x00\x00\x00\x04\xFE\x11\x00\x00@\xF9\v\x00\x00\x00 \xF6\xFA\x00\x00\x00\xE0\x00\xFE\x00\x00\x00\x80\xF7\xF0\x00\x00@\xE5\xFF\x00\xF8\x00\x00\xF8\x00\x0F\xFE\x01\xFE\x00\xE4"
```
But as i don't know what the communication is supposed to look like i don't know if that helped at all.

Are the UART answers as you would expect ?
I think the cause of the issue is, that i am using a ESP32 and not a ESP8266, but i do not like to change that, because i have never used a ESP8266.

JonathanTreffler, 2022-03-25:
Does the default firmware of prebuilt tindie modules have UART included ?
The more i debug this it just seems like nothing but interference gets recieved by the esp.

JonathanTreffler, 2022-03-25:
More logs (no logic level invertion), but logging as hex, not strings:

```
[00:25:00][D][button:013]: 'Desk Position 2' Pressed.
[00:25:00][D][uart_debug:114]: >>> 3C:4C:30:2C:32:2E
[00:25:00][E][uart:015]: Reading from UART timed out at byte 0!
[00:25:00][D][uart_debug:114]: <<< 58:3B:00:68:FE:81:00
[00:26:02][D][button:013]: 'Desk Position 3' Pressed.
[00:26:02][D][uart_debug:114]: >>> 3C:4C:30:2C:33:2E
[00:26:08][D][button:013]: 'Desk Position 4' Pressed.
[00:26:08][D][uart_debug:114]: >>> 3C:4C:30:2C:34:2E
```

tagno25, 2022-03-26:
Odd that your logs are being printed as hex, possibly a bug with the ESP32 build.

Unless @gcormier changed it since the v2021.09 build, serial is not enabled on the shipping boards.

If you don't have any programmer to flash a different firmware onto the megadesk controller, then you could try using [https://github.com/ESP32-Musings/OTA_update_AVR_using_ESP32](https://github.com/ESP32-Musings/OTA_update_AVR_using_ESP32).  I have not tried or used it, but it may be worth a shot before purchasing a programmer.

gcormier, 2022-03-26:
Phil is correct - the serial firmware is not flashed by default.

JonathanTreffler, 2022-03-26:
What are the reasons for not compiling it in the default firmware ?

The solution by @tagno25 requires bidirectional voltage shifters, which i don't have at my current location :/

Probably due to the chip shortage i can only find knock-offs of the AVR Programmer recommended in the DIY section for twice  the price in germany😐

So i guess my Megadesk project needs to be delayed about 6 months when i am back at this location.

Thanks anyways :)