gcormier/megadesk #107: ATTiny1624 Support

gcormier, 2022-08-29:
ATTINY1624 branch up in case the availability (or pricing) of this becomes better then ATTINY841.

Pros
- Two hardware serial
- Might be cheaper?

Cons
- UPDI programmers less prevalent

Need to work on code.

gcormier, 2022-08-29:
@philwmcdonald Unsure if any of the optimizations you made are related, I've only just flashed the first test and noticed the pitch is higher on the startup tone. Later this week I'll poke around. I'll probably just also flash a 4 line program that outputs a given frequency tone and measure it with the scope to see if it's something at the core of my configuration.

gcormier, 2022-08-30:
Woops, I thought I read the fuses set the default clock to 16MHz. The 1624 runs at 20MHz by default. A fuse change gets us at 16MHz to save a bit of power.

gcormier, 2022-08-30:
[image]

Serial oddities on LIN bus. Top is 1624, bottom is tiny841. No code changes.

philwmcdonald, 2022-09-02:
Never found a need for all that eeprom previously.
You could just initialize/use 256b instead without any impact to the operation of megadesk.

gcormier, 2022-09-10:
Bingo, setting pinMode fixed it!

gcormier, 2022-09-10:
Serial working well on 1624

[image]

gcormier, 2022-09-10:
Tested build against attin841, fixed some issues. Working!

closed 2022-09-10 by gcormier