gcormier/megadesk #133: Feature request "Silent mode"

gcormier, 2023-04-08:
### Discussed in https://github.com/gcormier/megadesk/discussions/132

<div type='discussions-op-text'>

<sup>Originally posted by **AnAnalogGuy** April  8, 2023</sup>
"Toggle Audio Feedback" does only supress audio feedback on pushing the buttons (at least on my megadesk), audio feedback from triggering megadesk to move into a specific stored position remains regardless of the "Audio Feedback" setting. 

While audio feedback is very useful in many situations and for many users, others might rather like a silent mode (me in example :-)). I was wondering if it would be an option to add such a silent mode?</div>

gcormier, 2023-04-09:
There's still quite a few sounds that could be controlled. For this new "silent" mode, do you still think having some audio when saving new memory positions would be useful? What about when starting up?

AnAnalogGuy, 2023-04-18:
Sorry Greg, i missed your question. 

Yes, i think it would be a good idea to have the sounds on as feedback to all user input in terms of programming, during startup, reboot and in error conditions. 

May thought comes from the perspective when controlling megadesk via ESP/software client. Which is moving the desk. 
In these situations there is no need for an audio feedback as from the software the user already knows/sees which action she/he has triggered. If i correctly get it, megadesk is silent when moving the desk to an abosult and a relative position, but it does beep when moving to a memory position. So i guess it's only the memory position which might also be included in the "toggle audio" condition so we have identical behaviour on all moving actions.

AnAnalogGuy, 2023-05-17:
Testing is still on my todo list, sorry for the delay.

webmediart-github, 2024-09-15:
> Testing is still on my todo list, sorry for the delay.

Any updates yet?

AnAnalogGuy, 2024-09-15:
> > Testing is still on my todo list, sorry for the delay.
> 
> Any updates yet?

You can test it if you like and let us know.