gcormier/megadesk #134: Megadesk recalibration throws error

buran-energia, 2023-04-17:
Megadesk hangs and then throws error when I try to recalibrate using it. Megadesk is controlled via Serial using ESP32 (24V).

Since I can only control it via Serial, I am sending the command "<L,14.", which should be equal to "14 UP pushes" that launches a recalibration procedure. When I do this, the desk goes down to the lowest position, but then Megadesk does not respond to any Serial commands. This isn't normal, right? Then upon switching power, it gives me ">E96,0" error no matter the command I send (e.g. "<C,0."). What does this error even mean? I can bring the desk back to life by using the original controller and moving the desk a bit up, but I prefer for recalibration to work via Megadesk.

Also I couldn't find anywhere in the docs: when do you need to recalibrate the motors? I assume it is when you get an error such as "DE4732,117" and the desk doesn't move to the correct preset position.

AnAnalogGuy, 2023-04-18:
> Also I couldn't find anywhere in the docs: when do you need to recalibrate the motors? I assume it is when you get an error such as "DE4732,117" and the desk doesn't move to the correct preset position.

Recalibration i.e. is needed when the two legs of the table are not moving in sync anymore, meaning, one leg reaches the lower or upper end earlier than the other.

AnAnalogGuy, 2023-04-18:
> Megadesk hangs and then throws error when I try to recalibrate using it. Megadesk is controlled via Serial using ESP32 (24V).

Are you using a ESP32 on 3.3 Volt? I'm asking, as I also from time to time have issues with megadesk and/or the ESP board crashing. Greg pointed out that the signaling Voltage of Megadesk is 5Volt which I guess is the root cause for the crashes i experience. Greg suggested to use a Voltage Level Shifter to make sure both boards are operating on 5V.

gcormier, 2023-04-18:
Can you remove your serial connection (TX,RX,GND) to megadesk and attempt the recalibration procedure to see if it works?

gcormier, 2023-04-18:
`>E96,0`
`>` start of comms
`E` error
`96` this is the target height it's trying to go to
`,0` memory position 0 (ignore)

96 is too low. It's probably via these lines https://github.com/gcormier/megadesk/blob/523fab165acd7ceec9f8d6c0d7a8b89f20e3d5c2/code/src/megadesk.cpp#L767-L773

We could change this to look at the direction of movement to error out accordingly. Eg if we are too-low for some reason, it could still allow UP movement.

buran-energia, 2023-04-19:
> Recalibration i.e. is needed when the two legs of the table are not moving in sync anymore, meaning, one leg reaches the lower or upper end earlier than the other.

So basically if it throws me a DE****,*** error, I need to recalibrate. Will this happen often?

> Are you using a ESP32 on 3.3 Volt? I'm asking, as I also from time to time have issues with megadesk and/or the ESP board crashing. Greg pointed out that the signaling voltage of Megadesk is 5V which I guess is the root cause for the crashes I experience (meaning the problem is not the megadesk). Greg suggested to use a Voltage Level Shifter to make sure both boards are operating on 5V.

I am using https://www.ezsbc.com/product/wifi01-sw/ as recommended on the site and haven't had any issues for a month now except the recalibration procedure. Does this thing need it? It is taking power from the same 24v source.

> Can you remove your serial connection (TX,RX,GND) to megadesk and attempt the recalibration procedure to see if it works?

It would be a pain to start recalibration, since I didn't open the original enclosure and don't have any buttons attached to megadesk. But I might connect some other button if it is really needed and there is such option.

How the recalibration should work, btw? Once megadesk lowers the desk, the recalibration is completed right away, and it should answer via serial?

> 96 is too low. It's probably via these lines

But that wouldn't explain why it becomes unresponsive upon recalibration, right?

gcormier, 2023-04-19:
> But that wouldn't explain why it becomes unresponsive upon recalibration, right, not even giving me E96 error until I restart it?

Correct, that is an odd one. I don't personally use serial control. But with just hardware, after recalibration, it will respond immediately.

Are you equipped to flash your megadesk with firmware?

gcormier, 2023-04-20:
You'll need an AVRISP to re-flash it, not a serial converter. A Pi will work as well in a pinch.

buran-energia, 2023-04-21:
> You'll need an AVRISP to re-flash it, not a serial converter. A Pi will work as well in a pinch.

Oh, right, I forgot I probably used RPi for that. I got RPi3.

gcormier, 2023-04-23:
In that case, if you are comfortable with it, you could flash some custom firmware that has different limits for the absolute min/max heights. If you are comfortable with `git` and vscode I can provide some guidance, otherwise I can give you a custom firmware to flash.

As for the AVR flashing itself yvia the Pi, you will need to rely on existing materials on the internet. I won't be able to assist much. ( https://github.com/gcormier/avr5hat/blob/main/code/pi_program.sh maybe be useful to you )

buran-energia, 2023-04-23:
> otherwise I can give you a custom firmware to flash.

That would be best. I will need to use a 3.3v-5v logic level shifter or what?

gcormier, 2023-04-23:
> That would be best. I will need to use a 3.3v-5v logic level shifter or what?

No, you can run megadesk at 3.3V and flash it. So just use the 3.3V from the Pi, MOSI/MISO and GND.

gcormier, 2023-04-23:
[megadesk-firmware-2023-04-23.zip](https://github.com/gcormier/megadesk/files/11305085/megadesk-firmware-2023-04-23.zip)

Here's a firmware you can try. It should allow you to move UP even if you are bottom'd out, or move down if you are at the very top.

This won't address the serial timeouts, but perhaps at least fix the other issue of being unable to move the desk ?

buran-energia, 2023-04-25:
Need some help with flashing before I brick something.

I connected it all like this: https://www.rototron.info/wp-content/uploads/PiSPI01.jpg

Added this to avrdude.conf:
```
programmer
  id    = "linuxgpio";
  desc  = "Use the Linux sysfs interface to bitbang GPIO lines";
  type  = "linuxgpio";
  reset = 12;
  sck   = 11;
  mosi  = 10;
  miso  = 9;
;
```
Did "sudo /usr/bin/avrdude -p t841 -c linuxgpio -v -t" and got "avrdude: AVR device initialized and ready to accept instructions". So it looks like the connections are fine.

Modified pi_program.sh:
```
#!/bin/bash

firmware=megadesk-limit_fix-t841-serial.hex

DEVICE=t841

LOW_FUSE=0xE2
HIGH_FUSE=0xD6
EXT_FUSE=0xFE

sudo avrdude -p $DEVICE -B 250khz -D -v -U hfuse:w:$HIGH_FUSE:m -U lfuse:w:$LOW_FUSE:m -U efuse:w:$EXT_FUSE:m #2>/dev/null
sudo avrdude -p $DEVICE -B 1MHz -v -U flash:w:$firmware:i #2>/dev/null

```

Can I run it? Or what else do I need?

gcormier, 2023-04-25:
You can remove the fuses since they will already be set properly. Fuses
would be the biggest risk when flashing.

buran-energia, 2023-04-25:
I flashed it, but now I get ">E0,0" when I try to recalibrate:
```
cmd: <L14,0.
echo: <L14,0
result: >E0,0
```

There was also some odd behavior. When I repluged power, it lowered the table to ~160 position from ~500.

buran-energia, 2023-04-25:
It also gives me the same error with "<L15,0", but I can move the desk with "=500.", so that works.

buran-energia, 2023-04-25:
(ignore the last 2 posts that I deleted, I issued wrong cmds).

Recalibration seems to work now. I flashed it, did "L,14.", it lowered the desk and didn't even hang.

gcormier, 2023-04-25:
So I just dug through some more of the code, and the edge case was already accounted for

https://github.com/gcormier/megadesk/blob/523fab165acd7ceec9f8d6c0d7a8b89f20e3d5c2/code/src/megadesk.cpp#L754-L765

So it is weird that with stock firmware, you were not able to "escape" out of the lower limit.

gcormier, 2023-04-25:
I don't even think it's possible, but I would be curious if you re-flashed the stock serial firmware if you have any issues. I verify the firmware after flashing all the units, so it should technically be impossible to have a "bad flash".

buran-energia, 2023-04-25:
> So I just dug through some more of the code, and the edge case was already accounted for

The code you linked seems to be related to manual move up and down, right? I don't have any up/down buttons connected to it and only control it using serial, remember? So after calibration, it would hang and after restart it would give me E96 error even for commands such as "C0,0" (get position). That was the problem.

buran-energia, 2023-04-25:
I did recalibration again, it lowered the desk to minimum, and this time it hanged (no serial responses). But I repluged power, and I don't get the E96 error anymore with this fix and was able to move the desk with serial. So the new flash seems to have somewhat fixed my initial problem.

buran-energia, 2023-04-25:
So hmm a new issue. Any idea why with this fix it sometimes lowers the table a bit upon pluging power? E.g. I plug the desk power, and the table goes down from 500 to 160 position for no reason. And this is despite setting the "lowest minimum" to 500 using "L,11." Any idea why it would move on its own now?

gcormier, 2023-04-25:
There might be some issues with the legs/motors. Already the recalibration should not be going as low as it is. You could try to use the MIN limit functionality to set the lower limit to be a bit higher the the absolute minimum to solve the issue.

buran-energia, 2023-04-25:
And a new one. It seems to have stopped remembering the positions I saved after the power is lost. E.g., I save positions using "<S500,2." and "<S1000,3.". Then unplug it for a few minutes, plug it back and I get a sad tone when trying to move to them. This wasn't a problem before flashing.

gcormier, 2023-04-26:
> So basically once you cut desk power for a bit and plug it in, it always
> lowers the desk to minimum position for no reason by itself. And it doesn't
> seem to remember anything from the eeprom (e.g. <"R,2." will read zero
> despite there being a saved position before replugging).
> There's some bigger issues. But if you mean setting the lowest limit using
> "L,11", it didn't work due to the eeprom issue I guess. I set it to 560
> (verified using R.11.). Then after replugging power it reads 299.

I'm still wondering if there's some serial hiccups at the start. Can you
unhook the serial control to see if that behaviour happens when powering on?

buran-energia, 2023-04-26:
> Can you unhook the serial control to see if that behaviour happens when powering on?

I unplugged all cables going to ESP32. Still lowered the desk by itself upon getting power.

gcormier, 2023-04-26:
Can you reach out to me via Tindie? Maybe there's a hardware issue.

buran-energia, 2023-04-26:
Flashed megadesk-v2022.09-t841-serial.hex. It doesn't move the table on startup, but still doesn't remember positions.

gcormier, 2023-05-12:
Faulty hardware

closed 2023-05-12 by gcormier