gcormier/megadesk #140: Lock butons

gcormier, 2023-07-08:
Add functionality to lock buttons. Include serial command to toggle it as well.

_Originally posted by @landonr in https://github.com/gcormier/megadesk/discussions/136#discussioncomment-6387247_

philwmcdonald, 2024-04-24:
A reminder that removing the grey physical key disables the keys. Clearly that's not under software control.

Disabling the button interrupt will of course do this, but then requires serial to re-enable.
Presumably the setting should be non-volatile so that power-cycle gets the buttons working normally again - without requiring serial?

wendevlin, 2025-01-21:
Hi,
I have no buttons anymore, so it would be cool to have virtual switches. 

Setting the height with the number input in home assistant is not perfect.
Is it possible to solder buttons to the megadesk board?

Any plans when the feature arrive?

How to flash the new firmware to the board?

gcormier, 2025-01-27:
If you have no buttons,

1. You can use companion for software control via REST API, web, or home assistant
2. You can solder your own buttons. The bottom of the PCB has two pads for UP and DOWN, and you use the GND pin on the 6-pin programming header. Connecting one of the buttons to GND will trigger that action.