gcormier/megadesk #147: Fresh install, megadesk only beeps whatever I do.

Extenuate0083, 2024-01-09:
I just installed my new megadesk but it keeps beeping. Both buttons make different sounds so I'm pretty sure I've hooked everything up correctly but I can't press 16 times on the up button because it only beeps. Nothing else happens.

Please help!

[image]

https://github.com/gcormier/megadesk/assets/132627587/619b17fa-4b2b-4dbb-8531-5060860f8458

gcormier, 2024-01-10:
16x is no longer needed in the newer boards.

Have you tried removing unplugging your desk from the wall until it's dead and plugging it back in?

Extenuate0083, 2024-01-10:
Yes! Several times.

gcormier, 2024-01-10:
Maybe it got damaged in transit - can you reach out to me on Tindie with your order #?

Extenuate0083, 2024-01-10:
Absolutely! Is there something else I could try to see if the problem is on
my side?