gcormier/megadesk #156: Unable to save new minimum height

mieky, 2024-02-19:
Hi! New Megadesk user here (v11a if I recall correctly).

My current lowest desk position is set at a few centimeters higher desk allows. I would like it to be set at the very bottom so I can save it to a slot, but seem unable to do so:
- when I click UP 14 times, the desk goes all the way down as expected
- when I click UP 11 times to set this as the lowest point, I only get a "disabled" beep

Curious as to why would storing the lower point be disabled? Any way I can get this lowest-possible position saved to a memory slot? Currently it will only ever as down as the second-to-lowest position.

gcormier, 2024-02-24:
Not sure I follow your use case.

If you want the desk to go as low as possible, why bother with the minimum height setting?

If you are looking to limit the maximum height, use UP 12x at the max height you'd like. Do not use lower limits, and the desk will inherently only go as low as possible.

(As for the memory at the lowest, I have moved my desk as low as possible, and saved it to a memory position and recalled it as well which worked without issue.)

mieky, 2024-03-06:
Thanks for the reply @gcormier, sorry for not being to explain my case well enough!

What I would like to have is store a memory position with the lowest possible configuration the desk allows – this my preferred working height when sitting. However, I don't currently know how to make that happen.

AFAIK I haven't set any minimum height for Megadesk, but the lowest position it allows me to go is a few centimeters higher than what I would get with the default Bekant controller. It seems as if there's an implicit lower limit I cannot reset, which is a couple centimeters above this position (e.g. 61cm, where I'd like 57cm). 

Any advice on how I could remove the lower limit altogether?

gcormier, 2024-03-06:
There is a lower safety limit hardcoded in the firmware. You can edit this value, recompile, and reflash the firmware if you'd like. 

https://github.com/gcormier/megadesk/blob/eada425798dad95a4d8fdc1375178ac9d3189996/code/src/megadesk.cpp#L143

mieky, 2024-03-07:
Ok, that's a pity. My bad for not being aware of this feature. Thanks anyway! 🙏🏻

closed 2024-03-07 by mieky