gcormier/megadesk #157: Computer won't detect companion

mimre25, 2024-02-20:
Hi - I'm trying to flash the companion, but my computer won't detect it. 

I'm holding the button why plugging it in but to no avail.

Any suggestion on how to debug this? I've tried multiple different USB slots and cable :/ 

This is the best I can get:

```sh
(esphome) ➜  esphome git:(master) esptool.py chip_id                    
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyACM0
/dev/ttyACM0 failed to connect: Could not open /dev/ttyACM0, the port doesn't exist

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

```sh
(esphome) ➜  esphome git:(master) esphome upload megadesk-companion.yaml
INFO ESPHome 2023.12.9
INFO Reading configuration megadesk-companion.yaml...
Found multiple options for uploading, please choose one:
  [1] /dev/ttyACM0 (USB JTAG/serial debug unit)
  [2] Over The Air (megadesk.local)
(number): 1
esptool.py v4.6.2
Serial port /dev/ttyACM0
ERROR Running command failed: Could not open /dev/ttyACM0, the port doesn't exist
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --port /dev/ttyACM0 --chip esp32c3 write_flash -z --flash_size detect 0x10000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/firmware.bin 0x0 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/bootloader.bin 0x8000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/partitions.bin 0x9000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/ota_data_initial.bin locally.
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
esptool.py v4.6.2
Serial port /dev/ttyACM0
ERROR Running command failed: Could not open /dev/ttyACM0, the port doesn't exist
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 115200 --port /dev/ttyACM0 --chip esp32c3 write_flash -z --flash_size detect 0x10000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/firmware.bin 0x0 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/bootloader.bin 0x8000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/partitions.bin 0x9000 /home/martin/Tools/megadesk/esphome/.esphome/build/megadesk/.pioenvs/megadesk/ota_data_initial.bin locally.
```

gcormier, 2024-02-20:
The error in dmesg is odd.

What OS is this? Is it a Mac?

mimre25, 2024-02-21:
It's Ubuntu 20.04.

Anyhow, I got it to work with another laptop - feel free to close this :slightly_smiling_face:

closed 2024-02-21 by gcormier