gcormier/megadesk #165: ECS+ Support Workaround

martenmatrix, 2024-05-04:
Do ROL ERGO ECS+ Controllers work with the fully assembled PCB?
[image]
It seems that the inputs and outputs are the same, however there is currently no PCB in between, which I could replace. Would it work to just solder the three outgoing pins of Megadesks PCB with the cables and to attach the "3-stripe thing" to the PCBs correct ingoing ports?

gcormier, 2024-05-04:
Can you get more information on whatever that plugs in to?

martenmatrix, 2024-05-04:
Thanks for the quick reply. Of course!

[image]
[image]
[image]
[image]
[image]
[image]

I hope that's enough information! :)

gcormier, 2024-05-04:
> Thanks for the quick reply. Of course!
>
> 5511CDD3-0F83-4563-BEA5-14BB7FAC5002.jpeg (view on web)
> 904F1E92-196B-4BA9-A72A-9445398D63CD.jpeg (view on web)
> 4399E9FB-D988-44E5-81A5-DB7BD5F281A5.jpeg (view on web)
> 8A863E78-B1D1-43C4-9AD4-9DC1485DA918.jpeg (view on web)
> A2C69B2E-CE90-4C82-9FFA-33D5EC9A47CF.jpeg (view on web)
> DDAC3787-E98A-4151-8DB0-674937324982.jpeg (view on web)
>
> I hope that's enough information! :)

What is the model of desk? Not clear as to why there is a 4 pin connector
when there's just 3 wires

martenmatrix, 2024-05-04:
I was baffled by that aswell. The model name is on that silver sticker in one of the pictures, it's **GF-470**. However, I'm currently unable to find it online, bought it ~3 years ago. The only thing I can send you is [the manual](https://hta.hettich.com/fileadmin/Hettich_Technical_Assistant/PDFs/2019_10_29_Anleitung_Change_Top_Eco_EN_DE.pdf).

gcormier, 2024-05-05:
I don't think it's compatible. This looks like the RODULF a bit, and it's likely if you open up that box, you'll find some high powered MOSFET's and it's actually sending the current to the motors instead of a signal via LIN bus.

martenmatrix, 2024-05-07:
That's unfortunate :(. However, thank you so much for replying so quickly!

closed 2024-05-08 by gcormier