gcormier/megadesk #171: Slow response time when releasing button while moving

matejcik, 2024-07-18:
Megadesk will take a long time after holding down the button to start moving.
I presume that's something to do with how short- vs long-presses are handled.

But when _releasing_ the button, it also takes a bit of time before the desk stops moving.
This makes it rather difficult to stop the desk at the exact right height, and to make minute adjustments when almost there.
(another possibility is that the height granularity is too rough for my tastes? but after twiddling with it for a little while, I don't think that is the case).

I would like the desk to stop moving immediately after releasing the button (or as soon as technically possible) -- like the original.

---

(otherwise thank you for this project, it's awesome)

gcormier, 2024-07-19:
I'm not going to look into this as some delay is needed. You can play with `CLICK_TIMEOUT` if you want.

AnAnalogGuy, 2024-07-22:
Why don't you work with memory positions? Once you found your fav positions, you can store them the table will go to that exact height.

matejcik, 2024-07-22:
> Why don't you work with memory positions?

it's _finding_ the position in the first place that is problematic

AnAnalogGuy, 2024-07-22:
I understood. However, it shouldn't take long to find the right position since after a couple of trials you should be able to compensate for the delay releasing the button a bit earlier. Once you got the position, you could store and recall it again. 

Btw, when using mega companion togehter with mega connect, you could even position the table using the UI, which would make it very easy to hit the right heigth.

closed 2024-09-14 by gcormier