gcormier/megadesk #182: Esphome deprecated custom compotents

unn, 2025-02-19:
See https://esphome.io/guides/contributing#a-note-about-custom-components 

Using esphome 2025.2.0, is now throwing compile errors for:

```
sensor:
  - platform: custom
    lambda: |-
      auto megadesk = new Megadesk(id(uart_desk));
      App.register_component(megadesk);
      return { megadesk->raw_height, megadesk->min_height, megadesk->max_height };
```

caveman99, 2025-02-27:
a quick fix was adding the snippet from https://github.com/robertklep/esphome-custom-component

gcormier, 2025-03-01:
Fixed.

If anyone far better at coding than I wants to tackle the code, send PR's at https://github.com/gcormier/megadesk_companion_component 

A lot of the stuff still using lambdas could be moved into code.

commit 687c5731 (2025-03-01): fixes #182
closed 2025-03-01 by gcormier